ArduPilot/ardupilot #14314: Sub: Add fully vectored frame

Williangalvani, 2020-05-07:
Users have been asking for a frame like these:
[image]

It should be straight forward to implement.

DeltaROV, 2021-11-08:
Hi Willian,

Ill be working on it very soon. I am also looking at making a frame.

I can make a simple one for the BR2 crowd open source with STEP file. I can also make it in my shop fairly quickly. Any ideas? 

I'll most likely change the power factor to 0.75 instead of 1.0 because this is what works best for my power supply.

So I might just add a constant value named **pfac** so it can easily be changed for those who want something different.

Do you think this is something you could add in QGC?

I am presently building an ROV using your standard heavy config called ALON.

So I was planning on calling this ALON 8VEC so you could call yours BR2 8VEC? I guess BR2 OMNI also has a ring to it :)

Cheers,
E.

ES-Alexander, 2021-11-09:
> I'll most likely change the power factor to 0.75 instead of 1.0 because this is what works best for my power supply.

It may make more sense to allow the full range, but set the limits for a given vehicle using parameters (as covered [here](https://discuss.bluerobotics.com/t/raspberry-and-pixhawk-shutdown-problem/10814/2)) :-)

> I was planning on calling this ALON 8VEC so you could call yours BR2 8VEC? I guess BR2 OMNI also has a ring to it

The frame name should be general purpose for ArduSub (e.g. the BlueROV2 just uses `SUB_FRAME_VECTORED`, which is shown in QGC as `BlueROV2/Vectored`). In this case I don't think we're actually making a vehicle like this, just including it as an available frame option in ArduSub (and presumably QGC), so no reference required to Blue Robotics.

Ideally the name should be generalisable. Since the number of thrusters in an omnidirectional vectored setup likely determines the directions somewhat, it may make sense to do something like `Omni8-6DOF`, or `Omni-Vectored8-6DOF`, or `Omni8-Vectored-6DOF`. Those then leave scope for expansion if someone wants to implement+include a minimal omni frame with only 6 thrusters (similar to [this drone](https://www.youtube.com/watch?v=BgqfVLgR2Bo) maybe?), or perhaps a larger one with up to 12.

DeltaROV, 2021-11-09:
Hello,

I will add it as: Omni8-6DOF

Your drone sounds like children screaming lol.

Any suggestion on frame?

Cheers,
E.

ES-Alexander, 2022-09-01:
A frame that [Blue Robotics has tested](https://github.com/Williangalvani/ardupilot/commit/df49134d79b47b2a3883215fe5e401c480b3429c):

```cpp
         add_motor_raw_6dof(AP_MOTORS_MOT_1,     -1,             -1,             1,              -1,                 1,                  1,              1);
         add_motor_raw_6dof(AP_MOTORS_MOT_2,     1,              -1,             -1,             -1,                 1,                  -1,             2);
         add_motor_raw_6dof(AP_MOTORS_MOT_3,     1,              1,              1,              -1,                 -1,                 -1,             3);
         add_motor_raw_6dof(AP_MOTORS_MOT_4,     -1,             1,              -1,             -1,                 -1,                 1,              4);
         add_motor_raw_6dof(AP_MOTORS_MOT_5,     1,              1,              1,              1,                  1,                  1,              5);
         add_motor_raw_6dof(AP_MOTORS_MOT_6,     -1,             1,              -1,             1,                  1,                  -1,             6);
         add_motor_raw_6dof(AP_MOTORS_MOT_7,     -1,             -1,             1,              1,                  -1,                 -1,             7);
         add_motor_raw_6dof(AP_MOTORS_MOT_8,     1,              -1,             -1,             1,                  -1,                 1,              8);
         break;
```

[image]
The larger opening of each thruster shroud is its front. The frame should work with off-cartesian angles anywhere between 0-90 degrees, as long as the vehicle is three-plane symmetric. Angles closest to 45 degrees provide the most balanced performance, whereas more tilt in a given direction can bias the vehicle towards lateral, forwards, or vertical motion.