ArduPilot/ardupilot #14319: Copter: altitude source is inconstant with EK2_ALT_SOURCE

KimHyungSub, 2020-05-08:
**Issue details**
As you know, EK2_ALT_SOURCE is primary altitude sensor used by the EKF. When I changed the altitude source from barometer to GPS, the EKF followed height from GPS as shown below.
[image]

On the other hand, I modified source code to make SITL executed with EK2_ALT_SOURCE = 2.
when I changed the altitude source from GPS to barometer, it did not follow the altitude from barometer as shown below.
[image]

**Version**
Master

**Platform**
[  ] All
[  ] AntennaTracker
[V] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
ArduCopter

**Hardware type**
SITL

**Logs**
1. EK2_ALT_SOURCE 0 --> 2
https://drive.google.com/file/d/11SCLwCdJrlfO49RjRSIUT2O2lbUR7vhV/view?usp=sharing

2. EK2_ALT_SOURCE 2 --> 0
https://drive.google.com/file/d/1KqXLLD95jjamB8lM5vXIeix5_RRd1AdA/view?usp=sharing

magicrub, 2020-08-10:
Are you saying that 0 -> 2 works without reboot and 2 -> 0 does not?

rmackay9, 2020-08-10:
@KimHyungSub,

Thanks for this report!  Do you think you could test with EKF3?  The next official release will use the EKF3 by default so there's more of an appetite to ensure it's correct.

KimHyungSub, 2020-08-10:
@magicrub Yes, that's correct!
@rmackay9 Sure. I will report the result after I test the EKF3.

KimHyungSub, 2020-08-11:
I just tested the EKF3. The result is the same as the EKF2 cases.

1) I used the following commands to test altitude source change from barometer to GPS.
param set AHRS_EKF_TYPE 3
param set EK2_ENABLE 0
param set EK3_ENABLE 1
param set EK3_ALT_SOURCE 2

[image]

2) I modified the following source code.
Changing "AP_GROUPINFO("ALT_SOURCE", 9, NavEKF3, _altSource, 0)" to "AP_GROUPINFO("ALT_SOURCE", 9, NavEKF3, _altSource, 2)" at libraries/AP_NavEKF3/AP_NavEKF3.cpp.

I also used the following commands to test altitude source change from GPS to barometer.
param set AHRS_EKF_TYPE 3
param set EK2_ENABLE 0
param set EK3_ENABLE 1
param set EK3_ALT_SOURCE 0

[image]

* Log
1) EK3_ALT_SOURCE 0 --> 2 (from barometer to GPS)
https://drive.google.com/file/d/18N5eIrC_KCnCXveJ4p1m_Jc-q02nWAzE/view?usp=sharing

2) EK3_ALT_SOURCE 2 --> 0 (from GPS to barometer)
https://drive.google.com/file/d/15W-wPF4laznQXpGLYKstXgp4Ff3icgoo/view?usp=sharing

rmackay9, 2020-08-11:
This is over my head so we probably need @priseborough to comment on this.

priseborough, 2020-08-11:
Both EK2_ALT_SOURCE and EK3_ALT_SOURCE as specified in the parameter description require a reboot after being changed. The behaviour if this is not done is undocumented.

KimHyungSub, 2020-08-11:
@priseborough 
Thank you for the feedback. So, would updating the document be the solution?

priseborough, 2020-08-14:
@KimHyungSub Which document are you referring to? The parameter descriptions in the code specify a reboot is required.

amilcarlucas, 2021-02-18:
Can this be closed?

priseborough, 2021-02-18:
Yes.

peterbarker, 2021-02-18:
Thanks @amilcarlucas , @priseborough

closed 2021-02-18 by peterbarker