ArduPilot/ardupilot #14339: Copter: Number of rotations of MAV_CMD_NAV_LOITER_TURNS does not rotate.

muramura, 2020-05-11:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

Because the value of the get_angle_total method is larger than the actual YAW value, the rotation stops in the middle and the process does not finish.
The rotation speed is 90 degrees per second.
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_auto.cpp#L1839

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2020-05-11:
Ok, so the issue here is that the vehicle is trailing the target point that is moving around the circle.  This means that the vehicle stops before it actually completes the circle.  I think that if it's given a bit more time (perhaps using a delay) it will reach the final target... so one possible solution may be to check that the vehicle is within the WPNAV_RADIUS of the target somehow.

muramura, 2020-05-11:
@rmackay9 san.
I want to rotate the copter N times, with WP in place. The radius of the circle is set to 0.

muramura, 2020-05-16:
In the following, the number of preceding rotations and the number of specified rotations are judged.
Therefore, I adjusted the ratio of the overturn by adding a value of 0.0 or more to the specified number of rotations so that it becomes the actual number of rotations.
At SITL, it got good.
I think it would be even better if they let you get the actual number of rotations.

https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_auto.cpp#L1839