ArduPilot/ardupilot #14342: Rally point logic possible broken

SergeyBokhantsev, 2020-05-11:
As per rally point design (if I understand documentation correctly) the RTL in the shown case should be performed to a rally point (blue line). But [as per implementation](https://github.com/ArduPilot/ardupilot/blob/70f445e7f1c4154762d3718458aebb1f7e0bd8aa/libraries/AP_Rally/AP_Rally.cpp#L160), looks like it will always select HOME (red path).

[image]

This may be a regression of https://github.com/ArduPilot/ardupilot/commit/36cc66f0f387f95f6d20dbb282859b24b9fe27a6

**Version**
4.0

**Platform**
[x] Copter

IamPete1, 2020-05-12:
The logic looks OK to me, do you have a log of this happening?

SergeyBokhantsev, 2020-05-12:
@IamPete1  Sorry my initial log of the flight where I noticed a possible incorrect behavior is lost. 

Look at the comment at line 160:
> // if a limit is defined and all rally points are beyond that limit, use home if it is closer
    if ((_rally_limit_km > 0) && (min_dis > _rally_limit_km*1000.0f)) {
        return false; // use home position
    } 

in fact, if all RPs are beyond the _rally_limit_km, this code does the "use home" but not "use home if it is closer"

IamPete1, 2020-05-12:
looks like the comment is wrong, it does as expected reading the LIMIT_KM param description. Comment should just be `// if a limit is defined and all rally points are beyond that limit, use home`

SergeyBokhantsev, 2020-05-12:
@IamPete1  I think the comment is correct, it matches the RALLY_LIMIT_KM description:
>If the closest rally point is more than this number of kilometers from the current position **and the home location is closer than any of the rally points** from the current position then do RTL to home rather than to the closest rally point.

So let's invert the bold part, and we have: if the closest RP is more than RALLY_LIMIT_KM and the home location is even farther then use this nearest RP. Does this looks correct?

IamPete1, 2020-05-12:
hum, i see what you mean now. So there is a inconsistency if nearest rally distance is closer than home but still outside the limit km. Currently it will go home.

Should be a easy fix in  calc_best_rally_or_home_location, just move the limit check there. Can you do a PR?

SergeyBokhantsev, 2020-05-12:
Looks like there is one more broken thing:
> If RALLY_LIMIT_KM is set to 0 then the closest rally point is **always** used
But currently "always" is not true: HOME will be used if it is closer.

I must say the description of RALLY_LIMIT_KM parameter is pretty complex to understand. Some statements are ambiguous/not clearly defined.  

So trying to understand the exact behavior of the rally points, that was initially designed (i.e. the way RPs are described on wiki, not as they currently implemented):
- RALLY_LIMIT_KM defines the radius of RP's absolute priority. If RTL triggered inside this zone - copter will return to RP no matter what.
- If the RTL triggered outside RALLY_LIMIT_KM of any RP then a condition step in: copter may return to HOME or to RP, depending of what is closer.
- If the RTL triggered outside RALLY_LIMIT_KM and RALLY_INCL_HOME=DoNotInclude then copter will return to the nearest RP, no matter how far it is.

@IamPete1 does these statements look true?

rmackay9, 2020-05-12:
I don't think the copter should ever return to a rally point if it is outside of RALLY_LIMIT_KM regardless of whether home is closer or not.  It's possible that Tridge may have a different opinion but I think it's slightly more intuitive to work this way.

SergeyBokhantsev, 2020-05-13:
@rmackay9 so you vote for the red path on the picture? (that's differs from the behavior described in wiki) 
Actually different users may want different behavior, depending of circumstances. From the other hand I believe the Rally Points are about safety in the first place. So the simpler/oblivious the logic - the better. To make logic simpler we need to reduce number of conditions.

 A. (RALLY_LIMIT_KM=0 just means unlimited radius)
1. if RTL triggered inside RP's RALLY_LIMIT_KM: always return to RP
2. otherwise always return to HOME
3. RALLY_LIMIT_KM=0 equals to unlimited radius, so (as per .1) copter will always return to RP (never to HOME)
4. We don't need RALLY_INCL_HOME

 B. (RALLY_LIMIT_KM=0 means a special mode)
1. with RALLY_LIMIT_KM > 0 
   1.1. if RTL triggered inside RP's RALLY_LIMIT_KM: always return to RP
   1.2. otherwise always return to HOME
2. with RALLY_LIMIT_KM=0
   2.1. Copter will always return to a closest rally OR home (whatever closest)
   2.2. If RALLY_INCL_HOME=0 then copter will ignore HOME and return to closest RP

rmackay9, 2020-05-13:
@SergeyBokhantsev, I think B is best and I think this is what we have already except for the RALLY_LIMIT_KM = 0 meaning no limit.  I'm sure we need to keep RALLY_INCL_HOME because some Plane users apparently don't want to return home.

Thanks for bringing this up.

SergeyBokhantsev, 2020-05-13:
@rmackay9 thanks, one more question. In "B" you can see that a combination of RALLY_LIMIT_KM > 0 and RALLY_INCL_HOME=0 makes no sense. So we either need to:
- if RALLY_LIMIT_KM > 0 then always ignore RALLY_INCL_HOME setting (perform B.1)
- if RALLY_INCL_HOME=0 then always ignore RALLY_LIMIT_KM setting (act like RALLY_LIMIT_KM =0 and perform B.2)

rmackay9, 2020-05-13:
@SergeyBokhantsev, I think RALLY_LIMIT_KM > 0 and RALLY_INCL_HOME = 0 makes sense.  I think this is the most common case for Plane users.

There has to be a final fallback though to return home.  So if it can't find a rally point for whatever reason it needs to return home.

SergeyBokhantsev, 2020-05-14:
@rmackay9 could you please clarify what you expect from this combination?
>I think RALLY_LIMIT_KM > 0 and RALLY_INCL_HOME = 0 makes sense. I think this is the most common case for Plane users.

rmackay9, 2020-05-19:
We discussed this on the call and although we didn't have any really strong opinions on which way it should be I think the decision was that we would leave the behaviour like it is in the stable releases and update the docs.  I..e we should not fly to a rally point if it's distance is > RALLY_LIMIT_KM.  instead the vehicle should fly home even if that means flying further.

SergeyBokhantsev, 2020-05-19:
@rmackay9 @WickedShell @IamPete1  thanks for discussing that!

>I..e we should not fly to a rally point if it's distance is > RALLY_LIMIT_KM. instead the vehicle should fly home even if that means flying further.

Agree, because this particular statement is simple and straightforward. You've stated this earlier in this thread and good it was confirmed on dev call. This part is how it is implemented currently.

But, unfortunately, as I've mentioned above the "red or blue path" is not the only problem in current implementation. There are some another problems. That's why I proposed the A and B behaviors. Currently it is nor A nor B, but:
[image]
(I even can't imagine how this could be described clearly in a text form)

For example, as a user I set RALLY_LIMIT_KM to a some value (and didn't change RALLY_INCL_HOME, it is 1 by default) and flying INSIDE a rally point.
Then RTL triggered. Where do you think vehicle will return?
The answer is: you cannot know for sure. It may return HOME or RP.

Flowchart of B option, that was preliminary agreed above as good one:
[image]

Well, I understand the rally point feature is not something innovative thus not very interesting to discuss)) But I believe this is critical because it is about safety. Hope this makes sense.