ArduPilot/ardupilot #14347: EKF3: primary lane changes on GPS glitch

rmackay9, 2020-05-12:
When using EKF3, the primary lane always changes as a result of a GPS glitch even though (I think) there's no advantage to switching lanes.

This can be reproduced in SITL by doing the following:

- param set EK3_ENABLE 1
- param set EK2_ENABLE 0
- param set AHRS_EKF_TYPE 3
- loiter
- arm throttle
- rc 3 2000 (to climb to 10m)
- rc 3 1500 (to level the vehicle)
- param set SIM_GPS_GLITCH_Y 0.0001  (to cause a GPS glitch of about 10m)

the message, "EKF primary changed:1" will be displayed.  If the glitch is then removed the vehicle will switch back to the first lane.

[image]

knudj, 2025-11-03:
I tried the commands from above in SITL, however got the error message:

LOITER> param set SIM_GPS_GLITCH_Y 0.0001
LOITER> Unable to find parameter 'SIM_GPS_GLITCH_Y'

From the parameter list I then found that GPS should have a number, however the command still does not work:

param set SIM_GPS1_GLITCH_Y 0.0001
LOITER> Unable to find parameter 'SIM_GPS1_GLITCH_Y'
param set SIM_GPS2_GLITCH_Y 0.0001
LOITER> Unable to find parameter 'SIM_GPS2_GLITCH_Y'

Unfortunately the old module "ekf" is not available anymore, that brought quite some insight into ekf affinity and lane switching.

ChatGPT and Copilot gave me some python code to check for lane switching, however many of the commands and parameters seemed to be obsolete. I even tried to find the function calcLaneScores() in AP_NavEKF3_core.cpp, however was not successful.

It would be great if someone would come up with a recent SITL test scenario for ekf affinity and lane switching, preferably for copter and plane :-)

rmackay9, 2025-11-04:
Here's a vehicle specific issue related to the AHRS/EKF's GPS glitch handling https://github.com/ArduPilot/ardupilot/issues/31402

knudj, 2025-11-05:
Hi again, just want to add findings from plane simulation.

Flying the standard CMAC-circuit and introducing a glitch as above results into:

waypoint 3
400
200
Flight battery warning
AP: AHRS: DCM active
AP: AHRS: EKF3 active

Although GPS1 and GPS2 are enabled the AHRS switches to DCM and then recovers to EKF3 again.

param show SIM_GPS*_ENABLE
SIM_GPS1_ENABLE  1.0         # Enable
SIM_GPS2_ENABLE  1           # Enable
SIM_GPS3_ENABLE  0           # Disable
SIM_GPS4_ENABLE  0           # Disable

I would have expected that at first hand EKF3 switches to the second IMU core (lane) with GPS2 that has no glitch.

**Just to make my point again:**

EKF affinity and lane switching certainly is an advanced feature and the standard user might confidentially rely that it is implemented in the right way. Nevertheless since the algorithm obviously is implemented differently for vehicles (and maybe even for ArduPilot versions) there needs to be a way to check and understand the algorithm.

**SITL is the perfect tool to check vehicle-vise for EKF affinity and lane switching :-)
It would be a nightmare to do this on the flying field :-(
There is urgent need for a handy monitoring tool coming with MAVProxy :-)**