ArduPilot/ardupilot #14351: SITL compile error in fedora 32

jinchengde, 2020-05-12:
# Bug report
fedora 32 use gcc version 10 and I have compile error for sitl

**Issue details**
'''../../libraries/AP_NavEKF2/AP_NavEKF2_MagFusion.cpp: In member function ‘void NavEKF2_core::FuseMagnetometer()’:
../../libraries/AP_NavEKF2/AP_NavEKF2_MagFusion.cpp:694:44: error: ‘H_MAG[16]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  694 |             KH[i][j] = Kfusion[i] * H_MAG[j];
      |                                     ~~~~~~~^
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors

Waf: Leaving directory `/home/jinchengde/Code/ArduPilot/ardupilot/build/sitl'
Build failed
 -> task in 'objs/AP_NavEKF2' failed (exit status 1): 
	{task 140494858164592: cxx AP_NavEKF2_MagFusion.cpp -> AP_NavEKF2_MagFusion.cpp.0.o}
 (run with -v to display more information)'''

_Please describe the problem_

**Version**
master

**Platform**
[ *] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

jancoow, 2020-05-28:
Neither works on arch. Neither does the docker image work.

acxz, 2020-07-05:
I can confirm this bug exists on Arch as well.

khancyr, 2020-08-10:
closed by https://github.com/ArduPilot/ardupilot/pull/13921

closed 2020-08-10 by khancyr

rkroshan, 2020-10-03:
The problem is happening due to the latest update of GCC 10.2.0 . Downgrade the packages of gcc-core, gcc-g++ and libgcc to 9.3.0-2 in cygwin environment . Then the problem will be resolved.

Thodoris-Evangelakos, 2020-12-27:
What if we're not using cygwin? Should we still downgrade to that specific version?

lirahc, 2022-07-08:
> The problem is happening due to the latest update of GCC 10.2.0 . Downgrade the packages of gcc-core, gcc-g++ and libgcc to 9.3.0-2 in cygwin environment . Then the problem will be resolved.

Hello @rkroshan, how I make that in my ubuntu (WsL)

```
sudo gcc --version 
sudo ...
```
And after that I dont know how make this