ArduPilot/ardupilot #14353: EKF2/EKF3: beacons not working in SITL (and maybe not working on real vehicles)

rmackay9, 2020-05-12:
EKF2 and EKF3's beacon support is broken in SITL and perhaps on real vehicles.  I suspect the changes to how the AHRS origin is being set are partially to blame (beacon's are the one sensor that can provide an origin).

In SITL (at least) there are two problems:

- the user must set the EKF origin by right-mouse-clicking on the map (this was not necessary previously)
- the vehicle moves wildly around on the map

To reproduce the problem in SITL:

- Enable EKF3 (EKF3 is the better choice for Beacons)
  - param set EK3_ENABLE 1
  - param set EK2_ENABLE 0
  - AHRS_EKF_TYPE 3
- param set EK3_GPS_TYPE 3
- param set GPS_TYPE 0
- param set BCN_TYPE 10
- param set BCN_LATITUDE -35.363261
- param set BCN_LONGITUDE 149.165230
- param set BCN_ALT 584
- param set BCN_ORIENT_YAW 45 (really?)

The vehicle should automatically appear on the map (but it doesn't) so if using MAVProxy, right-mouse-button-click and "Set Origin (with Altitude)" and the vehicle will appear but then begin moving wildly.

To resolve the issue we should probably confirm that the [AP_Beacon_SITL](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Beacon/AP_Beacon_SITL.cpp) class is calculating reasonable distances.  This could be done without enabling the EKF to use the Beacons so we can maintain control of the vehicle (i.e. leave GPS_TYPE = 1, EK3_GPS_TYPE = 0).

The next step would be to confirm these values are making it into EKF3.  Finally we should investigate what's wrong with setting the origin, it's reasonable to remove the BCN_LATITUDE, LONGITUDE and ALT parameters though because this method of setting home is inconsistent with other sensors.

PR https://github.com/ArduPilot/ardupilot/pull/14345 partially addresses an issue with the Beacon autotest but does not fix the issues mentioned above.

peterbarker, 2020-05-12:
>     * the user must set the EKF origin by right-mouse-clicking on the map (this was not necessary previously)

I'm not sure this is correct.  Certainly the autotest isn't setting an origin with a mavlink message, yet it is definitely getting an origin:
```
pbarker@bluebottle:~/rc/ardupilot(pr/fix-beacon-test)$ mavlogdump.py /home/pbarker/rc/buildlogs/Copter.BeaconPosition-00000004.BIN --t ORGN
1970-01-01 10:00:40.44: ORGN {TimeUS : 40446315, Type : 0, Lat : -35.3629376, Lng : 149.1650816, Alt : 584.0}
1970-01-01 10:00:40.44: ORGN {TimeUS : 40446315, Type : 1, Lat : -35.3629398, Lng : 149.1650836, Alt : 584.0}
1970-01-01 10:00:42.31: ORGN {TimeUS : 42316400, Type : 0, Lat : -35.3629376, Lng : 149.1650816, Alt : 584.0}
1970-01-01 10:00:42.31: ORGN {TimeUS : 42316400, Type : 1, Lat : -35.3629387, Lng : 149.1650839, Alt : 583.99}
```

>     * the vehicle moves wildly around on the map

Yep, autotest definitely shows that.  I'm not chasing it, but I reckon it's a yaw thing.

> The vehicle should automatically appear on the map (but it doesn't) so if using MAVProxy, right-mouse-button-click and "Set Origin (with Altitude)" and the vehicle will appear but then begin moving wildly.

This shouldn't be required.  Would be interesting to know if the tlog is getting a `GLOBAL_POSITION_ORIGIN` message before you manually set the origin.

peterbarker, 2020-05-13:
So playing with that beacon yaw parameter (the one suggested be set to 45) yields some interesting things.

This is yaw-180:
[image]

yaw 90 is closer in position - but we're supposed to be tracking a straight line...

[image]

rmackay9, 2020-05-13:
I would guess that the yaw should be set to zero.. I'm slightly amazed that we would put the effort into the autotest to handle an orientation of 45deg.

peterbarker, 2020-05-13:
I doubt the 45 is correct.

Yaw-0 gives the same pattern as yaw-180 - not rotated 180, the same.

rmackay9, 2020-05-13:
maybe yaw-45?

peterbarker, 2020-05-13:
Argh!  I've been looking at the wrong logs.

Fixed test pushed.

[image]

yaw-0 was correct.

rmackay9, 2020-06-09:
Paul Riseoborugh passed on some information on issues with the Beacon code in the EKF:

- we need to ignore beacons if we are very close to them
- when the beacons are organised in a plane and the vehicle is close to the plane we cannot accurately calculate an altitude

rmackay9, 2020-08-31:
This is fixed in master so closing

closed 2020-08-31 by rmackay9