ArduPilot/ardupilot #14358: Building Docker image on Raspbian ( armhf ) fails

Glagnar, 2020-05-13:
## Bug report

**Issue details**

Building docker image on raspberry pi fails with an error:
Package 'g++arm-linux-gnueabihf' has no installation candidate

**Version**
4.0.3 Release Candidate

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Logs**
[image]
ow the issue_

khancyr, 2020-05-13:
hello, thanks for the report, I will correct it. 
It comes from the fact that this toolchain doesn't exist on raspi as it is the default one.
As temp fix, I think you could remove it from the Tools/environemnent_install/install-prereqs-ubuntu.sh .

I will try to have a backport of the new dockerfile on the copter4.X serie to correct some other mistake

Glagnar, 2020-05-14:
> hello, thanks for the report, I will correct it.
> It comes from the fact that this toolchain doesn't exist on raspi as it is the default one.
> As temp fix, I think you could remove it from the [...]

I removed the references from the file, and now it works on arm. Thanks.

khancyr, 2020-05-14:
yep the old dockerfile isn't great, that is why I will try to backport the new version into the 4.X branch

peterbarker, 2020-11-24:
@khancyr any movement on this one?

IamPete1, 2021-08-08:
sounds like this issue is resolve on the newer version of the dockerfile, so I'm going to close.

closed 2021-08-08 by IamPete1