ArduPilot/ardupilot #14360: Build Fails for SITL (macOS)

harshitsankhla, 2020-05-13:
## Bug report
After rebasing my fork. There were changes added yesterday (12th May) in **AP_Scripting/generator/src/main.c** 
Getting the same error with a fresh clone of the upstream repo as well.

**Commands followed**
git clone <...ArduPilot/ardupilot.git>
cd ardupilot
git submodule update --init --recursive
./waf configure --board sitl
./waf copter

**Error**
../../libraries/AP_Scripting/generator/src/main.c:683:14: error: variable 'operation' is used uninitialized whenever 'if' condition is false [-Werror,-Wsometimes-uninitialized]
  } else if (strcmp(operator, "/") == 0) {
             ^~~~~~~~~~~~~~~~~~~~~~~~~~
../../libraries/AP_Scripting/generator/src/main.c:689:28: note: uninitialized use occurs here
  if ((data->operations) & operation) {
                           ^~~~~~~~~
../../libraries/AP_Scripting/generator/src/main.c:683:10: note: remove the 'if' if its condition is always true
  } else if (strcmp(operator, "/") == 0) {
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../libraries/AP_Scripting/generator/src/main.c:676:3: note: variable 'operation' is declared here
  enum operator_type operation;

Build failed
 -> task in '/Users/harshit/git/ardupilot/build/sitl/gen-bindings' failed (exit status 1): 
	{task 4441728464: /Users/harshit/git/ardupilot/build/sitl/gen-bindings main.c -> gen-bindings}
 (run with -v to display more information)

**System Info**
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Using toolchain                          : native 
Checking for 'g++' (C++ compiler)        : not found 
Checking for 'clang++' (C++ compiler)    : /usr/bin/clang++ 
Checking for 'gcc' (C compiler)          : not found 
Checking for 'clang' (C compiler)        : /usr/bin/clang 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
Checking for feenableexcept              : no 
Disabling SLP for clang++
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : not found 
Checking for header byteswap.h                 : not found 
Checking for HAVE_MEMRCHR                      : no 
Checking for program 'python'                  : /usr/local/opt/python/bin/python3.7 
Checking for python version >= 2.7.0           : 3.7.6 
Checking for program 'python'                  : /usr/local/opt/python/bin/python3.7 
Checking for python version >= 2.7.0           : 3.7.6 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : enabled 
Scripting runtime checks                       : enabled 
Checking for program 'rsync'                   : /usr/bin/rsync

**Platform**
[#] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

harshitsankhla, 2020-05-13:
Compiles fine on ubuntu 18, seems to be an os/compiler specific issue

WickedShell, 2020-05-13:
Should be fixed with #14362 now merged, feel free to reopen it if it still happens.

closed 2020-05-13 by WickedShell