ArduPilot/ardupilot #14364: Copter: I can't write or read the automatic flight plan.

muramura, 2020-05-13:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

I am using APM PLANNER2 and Mission Planner.
I'm using APM PLANNER2 and Mission Planner to create my plan.
Up to AC3.6.12, both plans could be written and read.
However, in AC4.0.X, APM PLANNER2 is not able to write and read the plan.

Has there been a change in the writing and reading procedures?
If there is a change, where do I look for it?

**Issue details**

_Please describe the problem_

**Version**

AC4.0.X

_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2020-05-13:
Yes, we've moved to a new protocol I think.  @peterbarker knows the details.

peterbarker, 2020-05-13:
No, rally points and fences moved to using the same protocol as missions.

I'll test here later today and see what I see.

peterbarker, 2020-05-14:
... testing will be delayed until I can compile APM Planner2 again.  I'm asking around in gitter on that front.

peterbarker, 2020-05-14:
I'm using master here and don't have any issues reading or writing waypoints with apmplanner2

Can you give me a specific git hash, please?

muramura, 2020-05-14:
@peterbarker san. Check and Comments thanks.
Software hash info:
APM PLANNER2 : 750a12ac
ArduCopter V4.0.0 (49693540) 

SITL, APM PLANNER2 is running on MAC OS X 10.15.4.

peterbarker, 2020-05-14:
On Thu, 14 May 2020, murata,katsutoshi wrote:

> @peterbarker san. Check and Comments thanks.
> Software hash info:
> APM PLANNER2 : 750a12ac
> ArduCopter V4.0.0 (4969354)
> 
> SITL, APM PLANNER2 is running on MAC OS X 10.15.4.

OK, so someone just opened a PR against APM Planner2 which could very well 
fix the problem you're seeing: 
https://github.com/ArduPilot/apm_planner/pull/1207

If that was you - great stuff.  If not - do you have the capacity to check 
if that PR fixed your problem?

... because using uninitialised stack space is very, very rarely a good 
thing :-)

muramura, 2020-05-14:
@peterbarker san.  
I've incorporated this PR. 
https://github.com/ArduPilot/apm_planner/pull/1207

I was able to do that when I didn't have an automatic flight plan written into my FC.
I was able to execute this automatic flight plan.

[image]

I then rebooted the APM PLANNER2.
I can't get the auto flight plan to show up on the screen.
I created a new automatic flight plan in this state and did a write operation.
I couldn't write in.

peterbarker, 2020-05-15:
Still unable to replicate here, sorry.  Perhaps if you send me a mission file which I can load using `wp load` in SITL and then fails to load in APMplanner2?

muramura, 2020-05-15:
@peterbarker san. 

I was able to confirm that the automatic flight plan appears at startup, read, refresh, write & read by doing the following.

I added the following to the message that sets MAV_COMP_ID_MISSIONPLANNER in target_component.
mission_type = MAV_MISSION_TYPE_MISSION

[image]

[image]

muramura, 2020-05-15:
I had the following problem.
FC is returning a message response with the system ID set to 0.
If the SYSID is 0 in the MAVLink message, it is a broadcast message.
The GCS can be connected to multiple aircraft.
When an FC receives a broadcast message, it must set its own identity to the SYSID.

[20200516 5:30:00.003 DEBUG] - HBD: Heartbeat started
[20200516 5:30:00.432 DEBUG] - Send requestall data streams when heartbeat restarts
[20200516 5:30:02.874 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:02.875 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:02.878 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:02.880 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:02.882 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:03.344 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:03.346 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:03.347 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:03.349 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:03.350 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:04.357 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:04.360 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:04.362 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:04.364 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:04.365 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:06.344 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:06.346 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:06.348 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:06.349 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:06.350 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:07.062 DEBUG] - Param: "STAT_RUNTIME" :  QVariant(int, 69910)
[20200516 5:30:08.380 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:08.383 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:08.384 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:08.386 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:08.388 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:10.362 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:10.364 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:10.366 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:10.367 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:10.368 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:12.347 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:30:12.350 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[... cut ...]
[20200516 5:36:08.226 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:36:08.227 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:36:08.228 DEBUG] - handleWaypoint() - Rejecting message, check mismatch: current_state:  0  ==  5 , system id  0  ==  1 , comp id  1  ==  1
[20200516 5:36:09.897 DEBUG] - HBD: Heartbeat stopped

peterbarker, 2020-05-16:
@muramura so we're agreed that this was an uninitialised stack variable, and that https://github.com/ArduPilot/apm_planner/pull/1207# ought to have fixed the problem?

muramura, 2020-05-16:
@peterbarker san..  Yes, I agree. Check and Comments thaks.
I am happy to use APM PLANNER2 in the future.

closed 2020-05-16 by muramura

muramura, 2020-05-16:
I've put up a related fix that needs to be fixed, an issue with automatic flight plan reading, writing and loading to AC4.0.0 or later that causes a communication timeout and does not complete.

https://github.com/ArduPilot/apm_planner/issues/1209