ArduPilot/ardupilot #14384: Matlab Simlink GSoC goals

IamPete1, 2020-05-17:
This is a issue to track the progress and goals of this summers GSoC project.

bnsgeyer, 2020-07-24:
@IamPete1  How are things going with your GSOC project?  Sorry that I've been out of touch.  what is your next step?

IamPete1, 2020-07-24:
@bnsgeyer Hi BIll, this plotting in MATLAB is that latest stuff. [Demo 1](https://youtu.be/m_cmR_wLe5c) and [Demo 2](https://youtu.be/3Q1a2skm17g ) The next step I was going to look at is hard coding the oscillation detection stuff and using it to implement a 'quick tune'.

bnsgeyer, 2020-07-24:
@IamPete1  Do you think you will be able to look at having a library or mode coded up in simulink that would interface with AP in the SITL. That way users could use simulink to develop their control laws or library and be able to test it with SITL without having to autogenerate code from matlab and compile?  I think that would be a great capability for developers that like to use simulink.

IamPete1, 2020-07-24:
@bnsgeyer Yeah, I can take a look, the tricky thing is pausing the vehicle code while talking to simulink. I'm not really sure the best way to approach that.

IamPete1, 2020-08-30:
GSoC is now complete, see [GSoC 2020 – Matlab and Simulink conclusion](https://discuss.ardupilot.org/t/gsoc-2020-matlab-and-simulink-conclusion/61049)

closed 2020-08-30 by IamPete1