ArduPilot/ardupilot #14388: Bad data in Hobbywing ESC parser

JacobCrabill, 2020-05-18:
I've been exploring support for Hobbywing's various ESC telemetry protocols, and while looking at the AP_Periph HWESC parser I noticed that there is at least one bug regarding temperature decoding.  

See this line here:
https://github.com/ArduPilot/ardupilot/blob/e2d6e70895b0420f807bdae548e9f127bf18c3a5/Tools/AP_Periph/hwing_esc.cpp#L109

Based on how the received `uint8_t` is converted into a temperature in degC, there are a number of entries in the lookup table which are inaccessible due to redundant `adc_temp` values.

I have no idea where this table originated, and I don't know what the "right" answer is, but this table is clearly incorrect.

peterbarker, 2020-11-24:
I think you're right, and it's pretty easy to fix.

Is it a worthwhile patch if the thing is generated?  We'll save a dozen bytes or so at the risk of complicating things later, I guess.

I've created a PR and we'll see if it gets bounced.

AliSeedg, 2024-02-14:
> I've been exploring support for Hobbywing's various ESC telemetry protocols, and while looking at the AP_Periph HWESC parser I noticed that there is at least one bug regarding temperature decoding. [...]

Can I ask you some questions?