ArduPilot/ardupilot #14391: No min/max checking for some SITL parameters

KimHyungSub, 2020-05-18:
## Bug report

**Issue details**

I found that assigning values to some SITL parameters, causing memory violations and crash of AP as shown in below.
- param set SIM_GPS_GLITCH_Y 100
- param set SIM_GPS_GLITCH_X 100
- param set SIM_ADSB_COUNT 1

[image]

The problem is that there is no any valid range for SITL parameters. I would like to recommend creating some valid ranges and checking  min/max values.

**Version**
Master

**Platform**
[  ] All
[  ] AntennaTracker
[V] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Copter

**Hardware type**
SITL

rmackay9, 2020-05-19:
As a side note I think it would be nice if we had parameter descriptions including valid ranges.

KimHyungSub, 2020-05-19:
If ArduPilot provides users with these descriptions and valid ranges, it will be really nice!