ArduPilot/ardupilot #14393: Error: Mag (0) bad radius xxx expected 222

profotoevideo, 2020-05-19:
Hello! I'm Brazilian, I'm setting up a "Tarot - XS690", I bought a 3dxr "The Cube Black (AKA Pixhawk 2.1)" and "Here + V2 RTK GNSS (M8P)", And I can't do the compass calibration. A message always appears saying: "Mag (0) bad radius xxx expected 222" Remembering that "xxx" symbolizes a number that always changes. What should I do to fix it? My thanks to anyone who can help.

tridge, 2020-05-19:
can you do a compass calibration with the parameter LOG_DISARMED=1 set, and then send the onboard log? I'd be interested in seeing why it is failing.
The "expected 222" means you are in an area of the world with a very low magnetic field strength

profotoevideo, 2020-05-19:
> can you do a compass calibration with the parameter LOG_DISARMED=1 set, and then send the onboard log? I'd be interested in seeing why it is failing.
> The "expected 222" means you are in an area of the world with a very low magnetic field strength

I tried again and failed. These are the messages that appeared.

19/05/2020 14:36:28 : PreArm: GPS speed error 1.7 (needs < 1.0)
19/05/2020 14:36:28 : PreArm: Compass not calibrated
19/05/2020 14:36:28 : PreArm: Compass not calibrated
19/05/2020 14:36:28 : PreArm: Hardware safety switch
19/05/2020 14:36:28 : PreArm: Throttle below Failsafe
19/05/2020 14:35:58 : PreArm: GPS speed error 1.7 (needs < 1.0)
19/05/2020 14:35:58 : PreArm: Compass not calibrated
19/05/2020 14:35:58 : PreArm: Compass not calibrated
19/05/2020 14:35:58 : PreArm: EKF3 Roll/Pitch inconsistent by 10 deg
19/05/2020 14:35:58 : PreArm: Hardware safety switch
19/05/2020 14:35:58 : PreArm: Throttle below Failsafe
19/05/2020 14:35:28 : PreArm: Compass not calibrated
19/05/2020 14:35:28 : PreArm: Compass calibration running
19/05/2020 14:35:28 : PreArm: EKF3 Roll/Pitch inconsistent by 12 deg
19/05/2020 14:35:28 : PreArm: Hardware safety switch
19/05/2020 14:35:28 : PreArm: Throttle below Failsafe
19/05/2020 14:35:22 : Mag(1) bad radius 359 expected 222
19/05/2020 14:35:11 : Mag(2) bad radius 352 expected 222
19/05/2020 14:35:11 : Mag(0) bad radius 331 expected 222
19/05/2020 14:34:56 : PreArm: Mag yaw error x=0.0 y=0.0
19/05/2020 14:34:56 : PreArm: Compass not calibrated
19/05/2020 14:34:56 : PreArm: Compass calibration running
19/05/2020 14:34:56 : PreArm: EKF3 Roll/Pitch inconsistent by 36 deg
19/05/2020 14:34:56 : PreArm: Hardware safety switch
19/05/2020 14:34:56 : PreArm: Throttle below Failsafe
19/05/2020 14:34:26 : PreArm: GPS speed error 1.6 (needs < 1.0)
19/05/2020 14:34:26 : PreArm: Compass not calibrated
19/05/2020 14:34:26 : PreArm: Compass not calibrated
19/05/2020 14:34:26 : PreArm: EKF3 Yaw inconsistent by 24 deg
19/05/2020 14:34:26 : PreArm: Hardware safety switch
19/05/2020 14:34:26 : PreArm: Throttle below Failsafe
19/05/2020 14:33:56 : PreArm: GPS speed error 1.2 (needs < 1.0)
19/05/2020 14:33:56 : PreArm: Compass not calibrated
19/05/2020 14:33:56 : PreArm: Compass not calibrated
19/05/2020 14:33:56 : PreArm: EKF3 Yaw inconsistent by 140 deg
19/05/2020 14:33:56 : PreArm: Hardware safety switch
19/05/2020 14:33:56 : PreArm: Throttle below Failsafe
19/05/2020 14:33:26 : PreArm: GPS speed error 1.2 (needs < 1.0)
19/05/2020 14:33:26 : PreArm: Compass not calibrated
19/05/2020 14:33:26 : PreArm: Compass not calibrated
19/05/2020 14:33:26 : PreArm: EKF3 Yaw inconsistent by 138 deg
19/05/2020 14:33:26 : PreArm: Hardware safety switch
19/05/2020 14:33:26 : PreArm: Throttle below Failsafe
19/05/2020 14:32:56 : PreArm: GPS speed error 1.2 (needs < 1.0)
19/05/2020 14:32:56 : PreArm: Compass not calibrated
19/05/2020 14:32:56 : PreArm: Compass not calibrated
19/05/2020 14:32:56 : PreArm: EKF3 Yaw inconsistent by 85 deg
19/05/2020 14:32:56 : PreArm: Hardware safety switch
19/05/2020 14:32:56 : PreArm: Throttle below Failsafe
19/05/2020 14:32:26 : PreArm: GPS numsats 5 (needs 6)
19/05/2020 14:32:26 : PreArm: Compass not calibrated
19/05/2020 14:32:26 : PreArm: Compass not calibrated
19/05/2020 14:32:26 : PreArm: EKF3 Yaw inconsistent by 46 deg
19/05/2020 14:32:26 : PreArm: Hardware safety switch
19/05/2020 14:32:26 : PreArm: Throttle below Failsafe
19/05/2020 14:31:56 : PreArm: GPS numsats 4 (needs 6)
19/05/2020 14:31:56 : PreArm: Compass not calibrated
19/05/2020 14:31:56 : PreArm: Compass not calibrated
19/05/2020 14:31:56 : PreArm: EKF3 Yaw inconsistent by 23 deg
19/05/2020 14:31:56 : PreArm: Hardware safety switch
19/05/2020 14:31:56 : PreArm: Throttle below Failsafe
19/05/2020 14:31:26 : PreArm: Compass not calibrated
19/05/2020 14:31:26 : PreArm: Compass not calibrated
19/05/2020 14:31:26 : PreArm: Hardware safety switch
19/05/2020 14:31:26 : PreArm: Throttle below Failsafe
19/05/2020 14:31:20 : Frame: QUAD
19/05/2020 14:31:20 : RCOut: PWM:1-12
19/05/2020 14:31:20 : CubeBlack 002D0037 50485010 20323450
19/05/2020 14:31:20 : ChibiOS: d4fce84e
19/05/2020 14:31:20 : ArduCopter V4.0.3 (ffd08628)

Hwurzburg, 2020-05-19:
sorry, Tridge wants the dataflash log, if you enabled LOG_DISARMED =1, it will log to the SD as long a power is connected....those are Ground Station messages, not the data log...you will have to download them from the Mission Planner Data Flash tab, or directly from the SD card...

profotoevideo, 2020-05-19:
Sorry, I'm learning. I uploaded the file to the wetransfer: https://we.tl/t-ryasxOyu7u
I ask you to be kind enough to check and thank you for your help, because here in my country (Brazil) there is no organized community to help.

igorsgcampos, 2020-12-05:
I am also from Brazil, approximately at -3, -60 and it also gave me this error, but it expected 250 instead. The compass in question was Solo's external compass. My calibration attempts were consistently getting values at the 300-310 range.

I wonder if SAA plays a role in this, since we are both from Brazil: https://en.wikipedia.org/wiki/South_Atlantic_Anomaly

igorsgcampos, 2020-12-06:
Here are the logs:

[Solo.zip](https://github.com/ArduPilot/ardupilot/files/5649079/Solo.zip)

The day was as follows:
I was using QGC.
At first I took off and noticed that the vehicle wasn't following a straight path to the first waypoint, I switched to stabilize attitude mode and manually landed the vehicle. The GCS also reported a couple times an inconsistency with the compasses by ~20 degrees, but it got cleared, I thought it was due to the sensors being initialized.
I concluded there was an issue with the compass and decided to re-calibrate it, but QGC kept reporting that the value of 300-310 was not acceptable and it expected a value below 250, this was for the external compass, the internal one was calibrating fine.
I got blister on my hand from the many times I rolled it by holding one arm and the opposite leg while yawing repeatedly. After several attempts to get it to work, I decided to just disable the external compass, but then QGC reported the vehicle was not ready for takeoff as if it did not hold a GPS lock, which was false.
Then, I noticed on the map that the vehicle illustration was not reflecting the reality. The arrow was consistently pointing +45 degrees on a clockwise direction than it was supposed to be. I tried to set the orientation of the compass manually, but noticed that it would use the autopilot orientation as it was internal, then I attempted forcing it to be considered an external one, and played around with the settings to no effect.
Finally, I decided to disable the supposedly good internal compass and leave the external one enabled. I realized the representation was now matching the direction of the vehicle, moved it around a few times to ensure it was OK and finally proceeded with the mission.

I am using ArduCopter 4.0.5 on a stock Solo, the only changes I made were cherry-picking https://github.com/ArduPilot/ardupilot/pull/15878 to 4.0.5 and altering SoloGimbal.cpp as it was ignoring roll input for DO_MOUNT_CONTROL (I will push a PR to fix this soon).

igorsgcampos, 2020-12-06:
The internal compass seemed very consistent, but presented this 45 degrees offset that I mentioned.

igorsgcampos, 2020-12-06:
I realized something weird. The log files I got from Solo through SSH are incomplete. Approximately only half the CAM messages were logged. I will download the logs from the autopilot later, but it takes a long time as the speed is only 10KB/s. Could the iMX be losing log messages due to the baud rate? I didn't change any logging settings or baud rate from CubeSolo defaults.

igorsgcampos, 2020-12-08:
[image]

IamPete1, 2021-07-18:
Closing, but feel free re-open or ask over on the forum.

closed 2021-07-18 by IamPete1