ArduPilot/ardupilot #14399: Copter-4.0.3：No such file or directory

bfYSKUY, 2020-05-20:
in cygwin: sim_vehicle.py -v ArduCopter -f singlecopter --console --map

[311/663] Compiling libraries/AP_RCProtocol/AP_RCProtocol_IBUS.cpp
[312/663] Compiling libraries/AP_RCProtocol/AP_RCProtocol_DSM.cpp
[316/663] Compiling libraries/AP_ROMFS/AP_ROMFS.cpp
[317/663] Compiling libraries/AP_RPM/RPM_SITL.cpp
../../libraries/AP_OpticalFlow/AP_OpticalFlow_HereFlow.cpp:10:10: A fatal error：com/hex/equipment/flow/Measurement.hpp：No such file or directory
   10 | #include <com/hex/equipment/flow/Measurement.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

peterbarker, 2020-11-24:
I believe this ought to be fixed now with @bugobliterator 's recent massive changes to CAN.

peterbarker, 2020-11-24:
Uh.  Except that 4.0.3 was explicitly mentioned....

However, I've just compiled for 4.0.3 on Linux here and it doesn't fail with that error.

I expect a `git submodule update --recursive --init` would fix the problem.

Please reopen if this is still an issue for you.

closed 2020-11-24 by peterbarker