ArduPilot/ardupilot #14400: GPS_RAW_DATA on F9P can cause 2.5s GPS data lag

tridge, 2020-05-20:
Using GPS_RAW_DATA=1 at 115200 baud with a F9P can cause a large lag to develop on the GPS, enough to cause the EKF to diverge. The UART is just trying to handle too much data.
We either need a healthy check, or we need to disable the RXM raw logging on F9P if negotiated baudrate is less than 230400.

Naterater, 2020-05-26:
Tridge, I think we can handle raw messagesat 5Hz as long as the NMEA messages are turned off. Those take up a lot of bandwidth. I can almost log a full constellation of RXM_RAWX messages at 115200 and 10Hz. I am successfully doing it at 10Hz with just GPS/Glonass.

I also support moving up the baud rate though; I can achieve full-constellation 20Hz reliably at close to 500,000.

tridge, 2020-08-24:
PR #15121 adds detection of this fault

Yups1337, 2020-09-17:
Any chance to add these messages for better PPK and geotagging?

**UBX-RXM-SFRBX**
Broadcast navigation data subframe 

**UBX-TIM-TM2** 
Time mark data 

The TIM-TIM2 message is used for camera feedback, each time EXTINT1 (or 2) is pulled low, the exact GPS time is recorded. 
Using RTKLib emlid for PPK gives an interpolated positions file with all events.

Nice tutorial by **rtklibexplorer**
[https://rtklibexplorer.wordpress.com/2018/10/26/event-logging-with-rtklib-and-the-u-blox-m8t-receiver/](url)

This works with ublox M8T, M8P, F9T and F9P.

Best regards form Germany
Yannick

peterbarker, 2020-11-24:
Closing this as I believe it was solved by moving to 230400

closed 2020-11-24 by peterbarker