ArduPilot/ardupilot #14405: Improvement to Obstacle Avoidance GSOC Goals

rishabsingh3003, 2020-05-20:
I am consolidating the list of already opened issues that I will be working on for this year's Google Summer of Code. It will be easy to track my progress from here, and I will link all my PR's here.  It would also be nice to receive any advice on how I could approach these.
The following five issues will be the first on my to-do list:

- [ ] https://github.com/ArduPilot/ardupilot/issues/13215: BendyRuler and Object Database should work in 3D

- [ ] https://github.com/ArduPilot/ardupilot/issues/11961: Resolve BendyRuler's hesitancy

- [ ] https://github.com/ArduPilot/ardupilot/issues/11565: Recovery from BendyRuler avoidance may impact the fence

- [ ] https://github.com/ArduPilot/ardupilot/issues/11601: Vehicles using Dijkstra's may pass too close to the fence

- [ ] https://github.com/ArduPilot/ardupilot/issues/7706:  Vehicles should back away from objects in normal Avoidance

After the above is done, I hope to come to this issue if time permits:
- [ ] https://github.com/ArduPilot/ardupilot/issues/11562: Add proximity sensor support to Dijkstra's

The Spline WP methods are being changed, therefore this issue will only be dealt with after the Spline changes are finalized: 
- [ ] https://github.com/ArduPilot/ardupilot/issues/12691: Dijkstra's should work with Spline waypoints

Any changes to the current plan will be updated here.

rishabsingh3003, 2020-08-26:
I am including  other minor improvements/ Bug fixes made to Avoidance or related so we can have a summary of all the changes that were made in this short period:
1. Reject obstacles near home (easier autonomous takeoff's) - https://github.com/ArduPilot/ardupilot/pull/15136
2. Bug fix with "Stop mode" never reaching the desired margin - https://github.com/ArduPilot/ardupilot/pull/14817
3. Check if readings are valid before sending it down to OA DB - https://github.com/ArduPilot/ardupilot/pull/14793
4. Check if origin is set before running OA- https://github.com/ArduPilot/ardupilot/pull/14607
5. Log simple avoidance as "SA" messages (As part of backing away PR)

rmackay9, 2020-12-18:
I think we can close this now 'cuz this successful GSoC project has completed, thanks!

closed 2020-12-18 by rmackay9