ArduPilot/ardupilot #14438: Modify autotest to notify Discord when run is complete

peterbarker, 2020-05-24:
## Feature request

We need to cut back on the number of tests we run on CI (so they finish faster, given we can't reasonably increase the resources allocated there).  However, we still want more tests overall.  The autotest results tend to get ignored a lot of the time.

A push notification to Discord (summarising improvements or regressions) may allow us to quickly notice and fix regressions that occur because we're not running all of the tests in the CI environment.

**Describe alternatives you've considered**
Continue to run more and more tests in CI.  We already take too long and randomly time out, which is no good.
Run fewer tests, catch fewer regressions.
Spend time optimising the time taken to run tests.

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

TunaLobster, 2021-02-02:
I poked around a little bit for this. I made my own discord server and added a webhook (I think we already have that bit.) Went over to Github side for my fork and added a webhook there (I think master already has this as well). I made a few random commits to one of my branches and watched what happened.

With the "Check suites" option, each workflow triggered a webhook message in the Discord chat. Just the Github Actions ones. I'm not sure about semaphoreci or Azure. I can seen success, cancel, and failure webhook types so far. I haven't found a clear way to wrap together all of the workflows that run together for a PR. It's is quite talkative on this branch and would drown out anything else we might want to read in the github_activity channel. More fine tuned control might only be possible from writing our own webhook script and have that run on a server when all the checks on a PR finish.

Duke2nuke, 2026-11-16:
Hi, I am Aarambh, a 3rd-year CSE student prepping for GSoC 2027, and I have been looking at the Python tooling around ArduPilot. I saw issue 14438 about autotest results getting ignored because long CI runs finish quietly, and a Discord notification with a short summary would help the team spot regressions sooner. My plan is to add a --discord-webhook option to autotest.py and post the pass/fail summary when a run completes, using only the standard library so no new dependency is needed. I also noticed PR 31900 tried this and was closed without merging, so I wanted to check first. Is this still free, and would a fresh attempt along these lines be welcome? Happy to open the PR once you confirm.