ArduPilot/ardupilot #14446: Fail to maintain attitude during Flip mode

swham86, 2020-05-25:
## Bug report

When I adjusted some parameters related to roll axis, it enabled AC to fail to maintain attitude during Flip mode and then crash.
This can be reproduced in SITL by doing the following:
- param set ATC_RAT_RLL_FF 0.432304
- param set ATC_RATE_R_MAX 11
- mode guided
- arm throttle
- takeoff 50
- mode alt_hold
- mode flip

[image]

To fix this issue, it requires AC to check low angular velocity in roll axis and high roll axis rate feed forward  before conducting Flip mode.

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[V] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Copter

**Hardware type**
SITL

**Logs**
https://drive.google.com/file/d/1Tf-PqOsHLhKpiw2zsTrHB9DyMHFpF6Jm/view?usp=sharing

rmackay9, 2020-05-26:
This is really a support issue at this point I think so it should be raised in the forums first.  https://discuss.ardupilot.org/c/arducopter/copter-40/155

Setting ATC_RATE_R_MAX = 11 means that the vehicle will roll at a maximum of 11deg/sec.  That means 32 seconds to do a full rotation.  The parameter includes suggested values of 360 for "Slow", 720 for "Medium" and 1080 for "Fast" so 11 is far outside the recommended range.

We could possibly increase the parameter descriptions so the lower end of the range is higher but it's always possible for the user to override any GCS warnings.

closed 2020-05-26 by rmackay9

swham86, 2020-05-26:
Thank you for quick response!
I saw the suggested values. If the lower end of the range is increased, it will be nice!