ArduPilot/ardupilot #14448: AP_Mount: Gremsy PixyU issues

rmackay9, 2020-05-26:
This is a list of known issues with ArduPilot's support of the Gremsy PixyU 3-axis gimbal ([wiki](https://ardupilot.org/copter/docs/common-gremsy-pixyu-gimbal.html))

- During RC Targeting (when the pilot is in control of the gimbal's angles) the gimbal maintains a constants earth frame heading.  I.e. the gimbal does not automatically keep pointing in the same direction as the vehicle.  Pilot's probably want to be able to choose whether the gimbal should keep a heading relative to the vehicle's heading or relative to the earth-frame heading 
- ROI points in the wrong direction (i.e. pan/yaw is incorrect) unless the gimbal is powered up facing North.  The gimbal itself does not have a compass so it does not know where true North is.  Instead it only knows its startup yaw (presumably it has low drift gyros) so we should calculate the difference between the gimbal's yaw=0 and true North and then add this to pan/yaw requests sent to the gimbal.  Some ideas that may work for capturing this yaw offset:

  - record the AHRS's yaw soon after startup.  This will likely be correct at first but it could become incorrect due to EKF yaw resets that happen after copter's climb 2m above home during takeoff (either manual or automatic)
  - send a target yaw of 0deg to the gimbal, then retrieve the actual angle from gimbal (you would expect this to become zero after a few seconds but it may not for some reason - I'm not sure why), the yaw offset is the difference between the vehicle's ahrs and zero or perhaps the reported yaw.

peterbarker, 2020-05-26:
Is the stab_pan parameterineffective in RC targetting, then?

rmackay9, 2020-05-26:
@peterbarker, the stab-pan parameter is apparently only used by the servo backend (I just checked) so I don't think that will help.  I think we probably need to introduce a couple of submodes into the RC targetting mode to allow the pilot to determine which behaviour they would like for 3-axis gimbals and then add an auxiliary switch to allow changing the behaviour in-flight.

peterbarker, 2020-05-26:
@rmackay9 I do like the sound of that behaviour on a switch.

Is there some reason that stab-pan parameter *should* be unique to the servo backend?  Perhaps it could be come a "_DEFAULT" parameter for this new switch behaviour?

rmackay9, 2020-05-26:
@peterbarker,

I think stab-pan is for a different purpose.  The stab-roll/tilt/pan parameters are there to say whether or not the gimbal can control its own roll/tilt/pan or whether ardupilot needs to run a controller to do it.

The new behaviour we need is for whether or not the pilot wants the yaw/pan target to rotate with the vehicle or not.  i.e. are the pilot's yaw/pan inputs to be interpreted relative to the vehicle's heading or relative to true north (aka earth-frame).

rrr6399, 2020-11-20:
Here is a post that I think is related to this issue
https://discuss.ardupilot.org/t/logic-in-storm32-mavlink-driver-appears-flawed-for-3-axis-gimbals-like-the-gremsy-pixy-u/64354

rrr6399, 2020-12-27:
I wanted to see what you all thought about adding these parameters to the AP_MOUNT.cpp class:

    // @Param: _PAN_REF
    // @DisplayName: Mount pan reference
    // @Description: Mount pan reference - 0:Local device coordinates, 1:relative to North
    // @Values: 0:Local,1:Global
    // @User: Standard
    AP_GROUPINFO("_PAN_REF", 23, AP_Mount, state[0]._pan_reference, 0),

    // @Param: _PAN_CONTROL
    // @DisplayName: Mount pan control support
    // @Description: Mount supports pan without yawing the copter
    // @Values: -1:Auto,0:No,1:Yes
    // @User: Standard
    AP_GROUPINFO("_PAN_CONTROL", 24, AP_Mount, state[0]._has_pan_control, -1),

The pan_control parameter allows the user to specify whether the pan feature is automatically discovered from the backend driver or explicitly specified. This approach allows for cases where copter yaw can be used even for cases where the gimbal actually supports panning independent of the copter. Sometimes the gimbal panning feature isn't desired. 

The _pan_ref parameter is used to specify whether the pan angle is relative to the neutral position of the gimbal or relative to North. It may not be needed if we assume that relative to neutral will always be used if the gimbal supports pan control and relative to North if it doesn't, but that assumption is limiting.  Alternatively, parameter 4, 5 or 6 of the msg_command_long for mav_cmd_do_mount_control could be used to specify the pan reference so the reference is explicitly specified.

Any feedback is welcome!

amilcarlucas, 2021-01-26:
@rrr6399 you may want to joint forces with @spark404 in #16433

amilcarlucas, 2021-09-21:
Some of this might have been fixed in the now merged #16433

rmackay9, 2022-06-16:
These issues have been resolved with the new gremsy driver. https://github.com/ArduPilot/ardupilot/pull/20915

closed 2022-06-16 by rmackay9

rrr6399, 2022-06-17:
@rmackay9 I looked at master and it looks like the Gremsy driver has pan control hard-coded to true. What if we need to prevent the gimbal from panning and would like the copter to yaw instead? (We have a payload that the Gremsy can handle, but there are cables that prevent turning.) The pan_control option allowed the user to specify the type of behavior they need. Is there another way to do this? (Use the Gremsy driver, but have the copter yaw rather than the gimbal pan.)

rmackay9, 2022-06-24:
@rrr6399,

How about if we allowed you to treat a 3-axis gimbal like a 2-axis gimbal by setting the PAN_MIN/MAX values to 0?  That would essentially say that the yaw can't move and so we could treat it like a 2-axis from them on.  That would be quite an easy fix I think.

rrr6399, 2022-06-27:
@rmackay9, but that wouldn't yaw the copter automatically if you "panned" the gimbal would it? It looks like the logic in the MAVLink handler checks the gimbal driver to see if it supports pan and if it doesn't, it commands a yaw of the copter. 

Of course, there is a work around where the standard yaw commands are used to pan the camera, but then the GCS has to know the type of gimbal used and then use different command sets.

rmackay9, 2022-06-27:
@rrr6399,

ok, so the suggestion is that if the pilot changes the gimbal to RC_TARGETTING mode that when he/she commands the gimbal to yaw the copter yaws instead?  I suspect I'm misunderstanding so maybe you could describe the situation and commands used by the pilot?

rrr6399, 2022-06-29:
@rmackay9 The case I was thinking about was for the DO_MOUNT command handling in the older GCS_Mavlink code: 
https://github.com/ArduPilot/ardupilot/blob/Copter-4.0/ArduCopter/GCS_Mavlink.cpp#L604

If the gimbal doesn't support pan then it yaws the copter. I imagine it'd be nice for RC_TARGETTING to work the same way, but that wasn't my primary case. The original Storm32 driver had the has_pan_control hard coded to false and then new Gremsy driver has it hardcoded to true.

All the controls are sent via the GCS in my case (programmatically or via joystick or keyboard).

> ok, so the suggestion is that if the pilot changes the gimbal to RC_TARGETTING mode that when he/she commands the gimbal to yaw the copter yaws instead? I suspect I'm misunderstanding so maybe you could describe the situation and commands used by the pilot?

rmackay9, 2022-06-30:
@rrr6399,

ok, I was thinking of changing the handling of DO_MOUNT_CONTROL so the vehicle does **not** yaw.  I don't use gimbals very much and I think I probably added this feature originally but it seems weird to me now.  Anyway, if you like it as it is then I'll leave it like that.  We don't have a lot of users of AP's gimbal drivers as far as I can tell so you're a precious breed! :-)

rrr6399, 2022-07-01:
@rmackay9 Just happened to get involved in projects requiring the Gremsy Gimbal. I've enjoyed working with them. 

I haven't merged my PAN_CONTROL parameter into the mainline yet. I'll submit a pull request for that. It looks like a different call is being made now in master though when the gimbal does not support pan
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L735
It looks like rate is being set rather than angle. 

The other issue that you addressed with the new upcoming messages was choosing between earth and local coordinates for pan angle. Ideally, setting pan to 0 deg would not yaw the copter to 0 deg N when pan is not supported. In my local branch I have some logic to address that by assuming 0 is local angle and then calculating what the yaw should be if  the gimbal didn't support pan. Will your current changes address that issue?

rmackay9, 2022-07-01:
@rrr6399,

Ok great, thanks.  Yes, we've enhanced Copter's yaw control to accept either angle or rate or both.  Perhaps the name could be improved but the set_yaw_angle_rate() call now takes two arguments, the first being angle, the second being rate.

I hadn't specifically thought about the 0-yaw=North issue but this seems like a good time to bring up that the current handling of do-mount-control is very weird.  If you have a 3-axis gimbal it treats the yaw as a body-frame (aka local coordinates) but if you've got a 2-axis gimbal it treats it as a earth-frame yaw and yaws the copter.