ArduPilot/ardupilot #14463: Auto-Selectable Battery Profiles

tenten8401, 2020-05-29:
## Feature request

**Is your feature request related to a problem? Please describe.**
I have several different multi-purpose quadcopters that use different cell counts when being used for different purposes. Whenever I swap the batteries I have to pull out a ground station computer, link up and load the parameters or go through a full autotune again if I don't have a ground station with me. I'd also need to change battery failsafe voltages every time.

**Describe the solution you'd like**
Betaflight and iNav have this cool solution that auto-detects the battery cell count from the total voltage on startup and loads one of three profiles that contains pids and other misc stuff. I think this should also include battery failsafe settings as it's tedious to change them every time. I believe you can do this with scripting currently but that's not very user friendly and I don't believe every flight controller supports it(?).
You could turn on your multirotor and if the startup battery voltage is in a range you specify it should be able to automatically switch profiles and if it's not in the range it will stay on the profile loaded before (or a user-specified default?). This shouldn't be all parameters just tuning / battery related ones. In the case the user doesn't want to have these auto switch for safety reasons it should be possible to change them on a switch or something.

**Describe alternatives you've considered**
Saving parameters from a ground station and loading them based on what battery I've got, but this is clunky and requires me to carry a ground station, which I don't always have with me.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[X] Rover
[  ] Submarine

IamPete1, 2020-05-29:
You could do this with scripting

tenten8401, 2020-05-29:
I'm aware but it's not user friendly at all and my autopilot only has 1 MB flash so I can't even use it.