ArduPilot/ardupilot #14467: AP_NavEKF2:  external navigation is using GPS/Mag noise parameters

chobitsfan, 2020-05-29:
External navigation data handling has been moved to AP_VisualOdom and its delay parameters is moved to VISO_DELAY_MS. However, ext nav data is stilling using GPS measurement noise parameter (EK2_POSNE_M_NSE) and mag measurement noise parameter (EK2_YAW_M_NSE). Maybe it would be better to add corresponding parameters in VISO_

rmackay9, 2020-05-29:
@chobitsfan, yes, I agree we should do this and Paul has said so as well.

One thing that's not clear to me though is what to do for sensors that can provide the error estimate.  Here are some options:

- Use the param values if they are non-zero and the sensors provides all zero values in the covariance matrix from the vision-position-estimate, vision-speed-estimate MAVLink messages
- Use the param values as minimums. i.e. if the sensor provides numbers worse than this we use them.
- USe the param values as overrides.  i.e. if the parameters are non-zero we use them but if zero then we use the sensor provided values.

rmackay9, 2020-05-29:
By the way, Paul and I also had a talk about how to convert the covariance in the vision-position-estimate message into posErr and attErr values to pass to the EKF.

Below has the index values from the covariance matrix.  posErr could be the cuberoot(i1^2+i7^2+i12^2) or MAX(i1,i7,i12).  Something similar could be done for the attErr using cells 16,19 and 21.

  | x | y | z | roll | pitch | yaw
-- | -- | -- | -- | -- | -- | --
x | 1 | 2 | 3 | 4 | 5 | 6
y |   | 7 | 8 | 9 | 10 | 11
z |   |   | 12 | 13 | 14 | 15
roll |   |   |   | 16 | 17 | 18
pitch |   |   |   |   | 19 | 20
yaw |   |   |   |   |   | 21

chobitsfan, 2020-05-29:
Thank you @rmackay9 

> Use the param values as fall backs in case the sensor provides all zero values in the covariance matrix from the vision-position-estimate, vision-speed-estimate MAVLink messages.

I think maybe it is easier for users to understand

> Use the param values as minimums. i.e. if the sensor provides numbers worse than the parameter values we use them.

I think this is the way how EK2_POSNE_M_NSE works now

https://github.com/ArduPilot/ardupilot/blob/adeb37581f003eb44b2cf1f097e289eade395eab/libraries/AP_NavEKF2/AP_NavEKF2_PosVelFusion.cpp#L591-L593

By the way. Maybe I can help with this issue. But it may cause some code conflicts with #14368 and #14404

rmackay9, 2020-05-29:
@chobitsfan,

Yes, I think you're right that we should use the param as a minimum.

If you wanted to take this on that would be greatly appreciated.  Let's get the two linked PRs in first and then move onto this.  txs again!

chobitsfan, 2020-06-03:
> posErr could be the cuberoot(i1^2+i7^2+i12^2) or MAX(i1,i7,i12)

Hi @rmackay9 I am a little confuse. because these two values are not equal. Should I use MAX(i1,i7,i12) for less computation? Thank you very much

rmackay9, 2020-06-03:
@chobitsfan, indeed, the MAX() method is more conservative and faster to compute while the cuberoot() method is more correct.  I'd say if we can do a cuberoot without a terrible CPU cost we should.

amilcarlucas, 2020-06-04:
@chobitsfan thanks, I need this feature :)

PR #14516 (2020-06-09): AP_VisualOdom: support covariance from msg and add pos, ang err parameters
fix #14467

flight log
[vis_pos_ang_err.zip](https://github.com/ArduPilot/ardupilot/files/4727267/vis_pos_ang_err.zip)

1. Optitrack motion capture system did not supply covariance, only mean error. So I only test parameters part. I do not send covariance from mavlink msg
2. I do not have Intel T265 and it is not tested.
closed 2020-06-09 by rmackay9