ArduPilot/ardupilot #14475: PWM Out 1-5 doesn't work with DShot on Kakute F7 Mini

tenten8401, 2020-05-31:
## Bug report

**Issue details**
When using my Kakute F7 Mini, DShot outputs 1-5 do not work at all even with SERVO_BLH_MASK set to 63. 6 does appear to work. I'm pretty confident it's a bug as the wiki says all 6 outputs are dshot capable and one of them seems to work.
[image]
[image]

**Version**
Copter v4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Kakute F7 Mini

**Logs**
[image]

rmackay9, 2020-05-31:
I'm glad the RCOut banner is included in this report and that it's useful.

rmackay9, 2020-06-01:
@tenten8401, any chance you could post an onboard log file?

tenten8401, 2020-06-01:
@rmackay9 Here you go
[log_0_UnknownDate.bin.zip](https://github.com/ArduPilot/ardupilot/files/4711202/log_0_UnknownDate.bin.zip)

andyp1per, 2020-06-02:
@tenten8401 what version of blheli?

tenten8401, 2020-06-02:
I believe BLHeli_S 16.7, but shouldn't matter because it does the banner thing regardless if an ESC is even plugged in or not.

tenten8401, 2020-06-02:
I should also add that none of the protocols work besides PWM, oneshot included (except for output 6)

tenten8401, 2020-06-02:
Hmm.. so I just re-flashed 4.0.3 fresh and all 6 outputs work with Dshot now.. It's something with my parameters because after applying them it goes back to not working again..
[image]

tenten8401, 2020-06-02:
Found the issue I think, as soon as I set FRAME_CLASS = 1 outputs 1-5 stop working, if I set FRAME_CLASS = 2 (or any other value) they all work as expected. If I had to guess it's because of 1-3 and 4-5 not sharing one output group.

IamPete1, 2020-06-06:
Can you try using PWM type to set dshot rather than the BLHeli mask. Once you set a frame class it auto assigns motors to outputs 1-4 (for a quad). This then means MOT_PWM_TYPE is applied. I suspect we have a conflict between the BLHMask and the PWM_TYPE.

tenten8401, 2020-06-06:
@IamPete1 Yeah you're probably right, I didn't even know MOT_PWM_TYPE existed.
[image]

peterbarker, 2020-11-24:
My reading here is that the issue was sorted.  Please re-open if I'm wrong.

closed 2020-11-24 by peterbarker