ArduPilot/ardupilot #14524: Too high Target PIDs when synthetic airspeed very different from real. Result - strong roll oscillation at launch.

hyperion11, 2020-06-05:
## Bug report

**Issue details**

Too high Target PIDs when synthetic airspeed very different from real. Result - strong roll oscillation at launch.

**Version**
4.0.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
ZOHD Dart 250G, Stock ESC, motor and propeller 5inch.
No airspeed sensor
**Hardware type**
Pixracer 

**Logs**
https://1drv.ms/u/s!AuFPVI_eDaZbhMcZ5_ePPpgAVt860A?e=lgvbX4
In the beginning 2020-06-03 20-19-14.bin
[image]
In this log GPS speed (and real speed, there is almost no wind) 5 times greater than Aspeed. 
Аfter the moment when the airspeed equals the GPS speed, oscillations gone. And during rest of flight not happens.
In another flight in the beginning 2020-06-03 20-29-39.bin
[image] i also receive roll wiggle, but smaller. GPS.SPD in this situation higher than CTUN.Aspd ~4 times. And at the end of log file was third launch.
[image]
At this launch GPS.Spd not so greater than CTUN.Aspd and here is no wiggle. (2 times higher)
So i think PID scaling interconnected to synthetic or real airspeed and in some condition this process can produce twitching.

peterbarker, 2020-11-24:
@hyperion11 we've got this PR underway: https://github.com/ArduPilot/ardupilot/pull/15810 which should help with the oscillations by clamping the airspeed estimate.

If you're in a position to test that - experimental! - code, then that would help.

hyperion11, 2020-12-10:
sorry, i can't reproduce this problem now :(

peterbarker, 2020-12-10:
OK, thanks for reporting back.  Hopefully we've fixed the problem - or we do when we merge the linked PR.

Poke us again if you reproduce.

closed 2020-12-10 by peterbarker