ArduPilot/ardupilot #14526: Ardurover4.0.0 compilation failes at AP_Compass_Calibration.cpp

q-wertz, 2020-06-05:
## Bug report

**Issue details**

When compiling the source code on an Raspberry Pi following [https://docs.emlid.com/navio2/common/ardupilot/building-from-sources/](https://docs.emlid.com/navio2/common/ardupilot/building-from-sources/) with Navio2 hat the compilation fails with error

```
../../libraries/AP_Compass_AP_Compass_Calibration.cpp: In member function bool Compass::send_mag_cal_report(const GCS_MAVLINK&):
../../libraries/AP_Compass_AP_Compass_Calibration.cpp:260:13: error: too few arguments to function void mavlink_msg_mag_cal_report_send(mavlink_channel__t, ...);
compilation terminated due to -Wfatal-errors
```

I think this is only at the 'Rover-4.0.0' tag. The 'Rover-4.0' version does not throw an error at that point but would be good to have the most recent tag be buildable/remove the 4.0.0 tag.

**Version**
Rover4.0.0 tag

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[x] Rover
[  ] Submarine

**Airframe type**
Rover

**Hardware type**
Navio2

**Logs**
see above

khancyr, 2020-06-05:
hello, 
you need to update your submodules : `git submodules update --init --recursive`

q-wertz, 2020-06-05:
I think I found out the reason:

Just followed the instruction without thinking:
```
git clone https://github.com/ArduPilot/ardupilot
git submodule update --init --recursive
git checkout Rover-4.0.0
./waf configure --board=navio2
./waf rover
```
-> described error

When doing it in this order:
```
git clone https://github.com/ArduPilot/ardupilot
git checkout Rover-4.0.0
git submodule update --init --recursive
./waf configure --board=navio2
./waf rover
```
it works...

I will report this to emlid...

Not any kind of expert in this but as far as I understand how it is set up at the moment I always pull the most recent version of the modules.
Wouldn't it be more convenient to put the working state of the modules with a release of e.g. Rover-4.0.0 as later changes to the code of the modules might not work with the Rover-4.0.0 code?

rmackay9, 2020-06-05:
Great that you got it figured out.

I think the Rover-4.0.0 release does have links to the correct versions of all the submodules.  It's just that this extra "git submodule update --recursive" is required to update them.  They don't update as part of the "git checkout".  I don't think we would want to duplicate all the code from the various projects we rely on into those directory.  I'm not sure if it's possible for us to make a "git checkout" also automatically update the submodules allow the "git submodule --update" to be skipped.

q-wertz, 2020-06-05:
Thanks for the help.

closed 2020-06-05 by IamPete1