ArduPilot/ardupilot #14534: Prevent entering AUTO mode if "Old" mission has waypoint that is very far away

Ryanf55, 2020-06-06:
## Feature request

**When planning a new mission, I would like to prevent the autopilot from starting the mission if the mission is far away**

If I have an old mission in the autopilot, it may be at a different field very far away. My current mission may require multiple batteries, so I don't want my missions to be reset as mentioned in #6383, #5743. 

**Describe the solution you'd like**
I would like an option of _max_waypoint_distance_. If switching to auto mode and the aircraft determines the distance to the next waypoint is greater than _max_waypoint_distance_, it should abort the auto mission and RTL. Additionally, an error message should appear on OSD for it. "Aborting Auto. Next waypoint is X meters away. Limit is Y", where X is the distance the waypoint is from the aircraft and Y is _max_waypoint_distance_. 

**Describe alternatives you've considered**
Because the missions I run require multiple batteries, I can't use MIS_RETAIN. I also considered a geofence, but the geofence would need to be very large. 

**Platform**
-> Anything running AP_mission lib
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[ x ] Rover
[ x ] Submarine

**Additional context**
Here's a screenshot and video of what happened today. 
https://youtu.be/Cr0vdrRC7OQ
[image]
[image]
[image]

robustini, 2020-06-06:
This could be interesting.

magicrub, 2020-06-06:
I would suggest this being an arming check, not an auto mode inhibit.

rishabsingh3003, 2020-06-07:
This was already asked in https://github.com/ArduPilot/ardupilot/issues/1348
I did have a try at it in my PR here: https://github.com/ArduPilot/ardupilot/pull/13308
But unfortunately it was one of my first PR's ever in ardupilot and it needs several changes before it can be merged in. I will have a try at fixing my old PR up unless someone else takes this.

Ryanf55, 2020-06-07:
> I would suggest this being an arming check, not an auto mode inhibit.

But then you can't fly at all if there is a mission stored even if you don't intend to run it.

Ryanf55, 2020-06-07:
> This was already asked in #1348
> I did have a try at it in my PR here: #13308
> But unfortunately it was one of my first PR's ever in ardupilot and it needs several changes before it can be merged in. I will have a try at fixing my old PR up unless someone else takes this.

Yes that would be fantastic to have someone work on it, but my point still stands about how we should prevent AUTO engagement, not a flight in general. 

If it was pre-arm check, you would always have to clear your mission after each flight, and while that would prevent the flyaway, it does cause a user to have to use a computer before their next flight at a different field regardless of whether they intend to fly an AUTO mission or not. 

If you think pre-arm is good, why not add it as a an option that can be disabled in pre-arm checks, but always if a user disabled the prearm check, then it would still prevent AUTO engagement. That seems to address both use cases. However, this is a different request than #1348, so either or both can be implemented.