ArduPilot/ardupilot #14535: I2C address changed - Sonar not detected

diyquad, 2020-06-06:
**Issue details**
I have 2 sonar, GY-US42, like the maxibiotic. Only wiring 1 for testing.
I have changed the adress of one of the sonar, following maxbiotic recommandations 
https://www.maxbotix.com/articles/095.htm#operation

When i put the decimal address in Mission Planner, reboot, sonar never detected; I know the address i put is correct:
Default I2C address: 224 in maxbiotic arduino -> E0 or 112 for RNGFND1_ADDR in mission planner 
New address: 222 in maxbiotic aduino -> 111 RNGFND1_ADDR in mission planner (should be this)
(i tryed several other I2C address/decimal combinaison with same result)

Arduino recognize the I2C at new address, and sonar works well; but as soon as i change address it's not recognize anymore by mission planner; 
 If i put it back to default I2C address and set in mission planner address at 0 (or 112) it work perfectly

Was thinking maybe to hard code some I2C address and recompile?

Thanks for your help/debug

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Mini Pix 1.0
Sonar GY-SU42 v2

Hwurzburg, 2020-06-08:
the driver has a default address of 70.... https://github.com/ArduPilot/ardupilot/blob/66cbed78f594bc5e03606b5bd50825aa6d8c2bdd/libraries/AP_RangeFinder/AP_RangeFinder_MaxsonarI2CXL.h#L7
so this is a bit strange on several counts...

diyquad, 2020-06-10:
maybe because its not the original maxbotix sonar?

peterbarker, 2020-06-19:
@diyquad are you able to test the PR I just created?  AFAICS the maxbotix devices ignored the address field in the rangefinder parameters.

PR #14636 (2021-01-19): AP_RangeFinder: maxbotix: use i2c-address parameter
Closes #14535

This is contrary to the description on the ADDR field; 0 does not disable the maxbotix sensor, it uses the default address (0x70).  But then, it seems a few other drivers ignore that bit anyway.
closed 2021-01-19 by tridge