ArduPilot/ardupilot #14544: F7 target Chibios build error

priseborough, 2020-06-07:
When building Plane master at commit c1de346150e4f91bbeac68715768b0b8e1cad6d7 for the CubeYellow or MatekF765-Wing F7 targets I get the following error. All submodule are up to date. I can build for CubeBlack and CubeOrange

96/103] ChibiOS: Compiling flash.c
../../libraries/AP_HAL_ChibiOS/hwdef/common/chibios_common.mk:233: recipe for target 'modules/ChibiOS/obj/flash.o' failed

In file included from ../../libraries/AP_HAL_ChibiOS/hwdef/common/chconf.h:31:0,
                 from ../../modules/ChibiOS/os/rt/include/ch.h:90,
                 from ../../modules/ChibiOS/os/hal/osal/rt/osal.h:32,
                 from ../../modules/ChibiOS/os/hal/include/hal.h:28,
                 from ../../libraries/AP_HAL_ChibiOS/hwdef/common/flash.h:18,
                 from ../../libraries/AP_HAL_ChibiOS/hwdef/common/flash.c:51:
./hwdef.h:45:28: error: expected declaration specifiers or '...' before numeric constant
 #define STORAGE_FLASH_PAGE 1
                            ^
../../libraries/AP_HAL_ChibiOS/hwdef/common/flash.c:131:15: note: in expansion of macro 'STORAGE_FLASH_PAGE'
 static_assert(STORAGE_FLASH_PAGE < STM32_FLASH_NPAGES,
               ^
../../libraries/AP_HAL_ChibiOS/hwdef/common/flash.c:132:15: error: expected declaration specifiers or '...' before string constant
               "STORAGE_FLASH_PAGE out of range");
               ^

tridge, 2020-06-07:
can you send me the generated hwdef.h from build/CubeYellow ?

priseborough, 2020-06-07:
Further to this I tried running install-prereqs-ubuntu.sh to update the build environment and got the following error:

dpkg: unrecoverable fatal error, aborting:
 files list file for package 'libwebkit2gtk-4.0-37:amd64' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)

I'm running Ubuntu 18.04LTS

peterbarker, 2020-06-07:
On Sat, 6 Jun 2020, Paul Riseborough wrote:

> Further to this I tried running install-prereqs-ubuntu.sh to update the build environment and got the following error:
>
> dpkg: unrecoverable fatal error, aborting:
> files list file for package [...]

That looks like an OS-screwed-up sort of an error.

Try re-running.

Peter

tridge, 2020-06-07:
the cause was using g++ 4.9, I'll fix the build with the old compiler and do a PR
meanwhile, we've moved paul to the 6.3 compiler

closed 2020-06-07 by tridge