ArduPilot/ardupilot #14546: IMU Values disappearing in Kakute F7 V1.5 (Non AIO)

NitinJSanket, 2020-06-07:
## Bug report
The IMU values seem to be disappearing for a small-time when using Kakute F7 AIO V1.5 (Non AIO) version. This makes the quadrotor twitch and sometimes flip and crash. The problem is not observed on the AIO version however. 

**Version**
4.0.3 and 4.0.4dev

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X4.

**Hardware type**
Holybro Kakute F7 V1.5 Non AIO.

**Logs**
https://drive.google.com/file/d/1rZ9QKi2Y5MBkVWBRDMVS9TXJD5PNpas7/view?usp=sharing

NitinJSanket, 2020-06-07:
Also been having same twitch issues with Kakute F7 Mini (cant log though).

IamPete1, 2021-08-16:
Is this still a issue on 4.1?

IamPete1, 2022-09-11:
No updates, so I'm going to close.

closed 2022-09-11 by IamPete1