ArduPilot/ardupilot #14559: AP_Bootloader Fails on Compilation

davidbitton, 2020-06-08:
## Bug report

**AP_Bootloader Fails to Compile**

I tried to compile a bootloader for the KakuteF7 board. I'm on a Macbook running Catalina. This is the error:

```
18:18:51 runner ['/usr/local/bin/arm-none-eabi-ar', 'rcs', 'Tools/AP_Bootloader/liblibcanard.a', 'modules/libcanard/canard.c.1.o']
../../Tools/AP_Bootloader/support.cpp: In function 'void flash_init()':
../../Tools/AP_Bootloader/support.cpp:95:21: error: comparison of unsigned expression < 0 is always false [-Werror=type-limits]
     while (reserved < FLASH_RESERVE_END_KB * 1024U) {
```

**Version**
I'm using the master branch.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
KakuteF7

**Logs**
_Please provide a link to any relevant logs that show the issue_

auturgy, 2020-06-08:
Can you try doing a git submodule update and see if it still fails?

Regards,

James

davidbitton, 2020-06-08:
OK. I performed the submodule command in order to clear a missing uavcan module. That cleared the missing module, but the compile failed with the aforementioned error(s). Just for kicks, I dropped the ardupilot folder and re-cloned along with a `submodule init --recursive --init`. I still receive that compilation error.

davidbitton, 2020-06-08:
As an add-on, I was doing some reading on the error. The error makes sense because
`FLASH_RESERVE_END_KB` is `#define`ed as 0.

auturgy, 2020-06-08:
I couldn't replicate.
What I did was:
git pull upstream master
./waf clean
./waf configure --board=kakutef7 --bootloader
./waf bootloader

this built successfully on my machine.
Which gcc version (gcc --version should return it).
My test might not be representative, as I'm currently testing with gcc 9 on my machine, not the default toolchain version.

davidbitton, 2020-06-08:
`gcc --version`
```
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.3 (clang-1103.0.32.29)
Target: x86_64-apple-darwin19.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```
As for git and waf, I'm following the same steps as you.

davidbitton, 2020-06-08:
this too:
```
➜  ardupilot git:(master) /usr/local/Cellar/gcc-arm-none-eabi/20160928/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

auturgy, 2020-06-08:
I'm really not familiar with MacOS and clang, so can't really say whether that is the problem or not.
You might need to force MacOS to compile with gcc?

andyp1per, 2020-06-09:
You need to use gcc 6 or later on the ARM side.

andyp1per, 2020-06-09:
Works for me on macOS with:

DB0837:ardupilot andypiper$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

davidbitton, 2020-06-09:
@andyp1per I noticed your `arm-none-eabi-gcc` is version 6.3.1 and mine is at 5.4.1. Is **[this](https://ardupilot.org/dev/docs/building-setup-mac.html)** how you setup your build environment? Thanks.

P.S. Turns out my iMac at work has version 7.3.1. SMH

andyp1per, 2020-06-09:
@davidbitton - no, I run ./Tools/environment_install/install-prereqs-mac.sh

davidbitton, 2020-06-09:
@andyp1per it looks like they wrapped up what was on that webpage into a Bash script.

IamPete1, 2021-08-08:
sounds like this was a compiler version issue, so I'm going to close.

closed 2021-08-08 by IamPete1