ArduPilot/ardupilot #14574: GPS_Auto_Switch

colibrex, 2020-06-10:
## Feature request

We are flying and measuring using high end RTK GPS. We use also the GPS as ALT_SOURCE for EKF2.
It happened that autopilot switched between our RTK to Here GPS while in flight which causes strong movement of the drone and loss of altitude. 
Reading the source code it looks like when the GPS have same status then the primary GPS becomes the one with more satellites. But it looks like our RTK has always less satellites. 

To avoid the unnecessary changes, we should have the possibility to define a primary GPS and to define under which condition the autopilot should use the second GPS. For instance in our case we still want to use the RTK GPS even if it loses FIX status. We would like to switch to here GPS only if the RTK GPS is not working at all. Or maybe to define a new mode in "GPS_AUTO_SWITCH" where it would switch only if primary GPS is dead.

[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Thank you, 
Ced

yanivasy, 2020-07-16:
I'm not sure that number of satellites is the criterion for selecting best GPS.
I think that solution type (single, RTK etc. ) is used primary.
Where in code you saw that?

IamPete1, 2021-08-16:
closed by https://github.com/ArduPilot/ardupilot/pull/15520

closed 2021-08-16 by IamPete1