ArduPilot/ardupilot #14581: Serial 1 and Serial 2 shared bandwidth issue

VulcanUAV, 2020-06-11:
## Bug report

NOTE this issue is apparent when using the cube 'orange' but does not show when using cube black.
Response from Peter Barker after he ran tests to identify if this was a genuine issue:

Huh. I seem to have replicated (the problem).

It looks like SERIAL1 and SERIAL2 are sharing bandwidth, with that
bandwidth defined by SERIAL1’s baudrate. Possibly a clock-sharing
problem.

If I configure SERIAL4 identically to SERIAL2 (and deconfigure SERIAL2),
then I get 4Hz on both links. If I have the same setup on SERIAL1/SERIAL2
they share bandwidth; at SERIAL1_BAUD=57600 SERIAL2 gets ~1.8Hz, at
SERIAL1_BAUD=38400 it’s sub-1Hz.

Sadly I can’t confirm the same is true on a CubeBlack on its standard
carrier board. My bench test setup has a faulty Telem2 port. I can say
that using SERIAL1/SERIAL4 works as on the CubeOrange - i.e. just fine.

So. Looks like we have an issue - could you raise one on our github repo,
please?

p.s.

I can set streamrates of 50 locally on SERIAL1/SERIAL4 and I get all
(relevant) streams at 50Hz, which is cool. Using a rather higher
buadrate :wink:

Interestingly, SERIAL1 caps out at 58 messages/stream/second, serial4 at
~113/stream/second.

**Version**
V4.0.3 on cube orange

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Multi copter X8

**Hardware type**
Cube orange

**Logs**
N/A

peterbarker, 2020-06-14:
OK, a hint from @tridge led me to investigate potential DMA sharing.

Telem1 and Telem2 do currently share DMA channels (as configured by its hwdef.dat).

Setting SERIAL2_OPTIONS to 512 (TX_NoDMA) allows Telem2 to meet the default 4Hz streamrate MAVPRoxy demands of it - without that option set it does not.

@proficnc for visibility

VulcanUAV, 2020-06-14:
Thanks Peter and everyone, will give this a go tomorrow!

Best Regards
Alex Hardy
Director
Vulcan UAV Ltd
+44 1989 555025
info@vulcanuav.com
------------------------------------------------------------------------

IMPORTANT - The information transmitted in this email, including any 
attachments, is intended only for the person or entity to which it is 
addressed. Such information may be confidential and privileged and no 
mistake in transmission is intended to waive or compromise such 
privilege. Any copying, review, re-transmission, dissemination or other 
use of, or taking of any action in reliance upon, this information by 
persons or entities other than the intended recipient is prohibited and 
could lead to prosecution. We will not accept any responsibility for any 
loss, damage or disruption caused to your data or computer system, which 
may occur while using the attached program(s) and/or data, whether as 
the intended recipient or in error. If you received this email in error, 
please delete it and inform info@vulcanuav.com immediately.

peterbarker, 2020-06-15:
Relevant https://github.com/ArduPilot/ardupilot/issues/14217

tridge, 2021-06-05:
fixed by #17663

closed 2021-06-05 by tridge