ArduPilot/ardupilot #14582: WatchDog reset in-flight

VDLJu, 2020-06-11:
## Bug report

**Issue details**
Autopilot was reset by a watchdog in-flight when executing a mission. Reset happened after executing the last mission item, which was a command, LOITER_WAIT for 10 seconds. Reset happened about few seconds after execution of LOITER_WAIT.

Following watchdog error line was logged:
[image]

Task: -2 if the fast loop had started
FL: Fault Line 100, the source code line number where the fault occurred.
FT: Fault type 3. 3 = Hard Fault (the most common)
FA: 404947019, Fault Address (in memory)
FP: 183, Thread Priority
ICSR : 4196355, Interrupt Control and State Register

**Logs**
[bin log pre WD reset](https://drive.google.com/file/d/1YgEK5JBXyH4OgPHrGN-udlRpIoWRaDwD/view?usp=sharing)
[bin log after WD reset](https://drive.google.com/file/d/12swkJ6oaPdOx5VFs_6xr7ETQpGD6Dkph/view?usp=sharing)
[Telemetry log](https://drive.google.com/file/d/1DJa22d0XCDRVRGxDFSn7nMIYMYsa9ipv/view?usp=sharing)

**Version**
ArduCopter 4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X4 copter

**Hardware type**
Cube black 

Previous discussion in the forum, link to it
https://discuss.ardupilot.org/t/crash-ac-4-0-3-watchdog-reset-in-flight-while-executing-a-mission/57652

rmackay9, 2020-06-12:
Thanks for the very detailed report including logs.

tridge, 2020-06-12:
@VDLJu I have looked at this quite carefully and unfortunately I don't yet have a clue as to the cause.
I notice a high baudrate on 921600 on telem1. Was there a companion computer attached? If so, do you have a record of the mavlink stream to/from the companion computer?

tridge, 2020-06-12:
the closest thing I have to a clue on this one so far is noticing the thread priority of 183 in the WLOG message. A priority of 183 means the monitor thread was running at the time of the fault. The monitor thread uses very little CPU (it sleeps almost all the time), so the fact it was running could be significant.
I did wonder if the stack size of the monitor thread, which is 512, is enough when there is a delay that triggers the MON logging. I setup a test to reproduce that and found it does have enough stack (about 192 bytes free when logging MON msg).
Right now the only guess I have is a nested interrupt happening during a MON message write causing stack corruption, but I can't prove that at all, and can't reproduce it

VDLJu, 2020-06-12:
> @VDLJu I have looked at this quite carefully and unfortunately I don't yet have a clue as to the cause.
> I notice a high baudrate on 921600 on telem1. Was there a companion computer attached? If so, do you have a record of the mavlink stream to/from the companion computer?

Unfortunately the companion computer doesn't log autopilot state. It mostly saves some external events, like image capture etc and does some realtime functions like translates mavlink to RC telemetry format. High baudrate is just there to minimize latency.

Let me know if I can help you somehow

VDLJu, 2020-06-12:
I thought that it's better to share some previous flights, in case these can give some insight to this mystery.

[A flight -1 prior the crash](https://drive.google.com/file/d/1oKF2Sp_K7cIgMoAUfSTNRI5Dwz5Tp0qE/view?usp=sharing)

[A flight -2 prior the crash](https://drive.google.com/file/d/1oKF2Sp_K7cIgMoAUfSTNRI5Dwz5Tp0qE/view?usp=sharing)

mmk0102, 2020-06-15:
FL: Fault Line 100, the source code line number where the fault occurred - can it give us some information?

rmackay9, 2020-06-16:
@mmk0102, yes, I think Peter and Tridge did use that information and narrowed down which line was last executed before the watchdog was executed but it wasn't clear how this could possible cause the problem.

peterbarker, 2020-11-24:
@tridge was this the DMA-teardown/setup race condition bug?

kumariitian121, 2022-06-08:
anyone working on this bug?

rmackay9, 2022-06-09:
@kumariitian121,

I suspect this particular watchdog has been fixed and this was on a pretty old version of AP (4.0.x).  Have you encountered a watchdog reset with 4.1.x or 4.2.0?

pliniocr, 2022-06-24:
There is some related with floating RX TX pins this problem?

pliniocr, 2022-06-24:
Is there any relationship with floating RX TX pins board?

amefabris, 2023-03-17:
I did, using FW 4.2.0 on a mroControlZeroF7, it gets triggered every time that I switch to LAND mode. I did a lot of flights with the same FW on a mroControlZeroH7 and I never encountered this issue...

peterbarker, 2026-11-15:
Closing as probably fixed: both suspected causes have been addressed. The monitor-thread stack is now 1024 bytes ([PR #14603](https://github.com/ArduPilot/ardupilot/pull/14603), [PR #20442](https://github.com/ArduPilot/ardupilot/pull/20442)), and the UART DMA transmit race that caused watchdog resets under high serial load was fixed in [PR #15525](https://github.com/ArduPilot/ardupilot/pull/15525). There have been no reports on current firmware; please open a new issue with logs if this recurs.

closed 2026-11-15 by peterbarker