ArduPilot/ardupilot #14584: Change in autotune parameter save-behaviour

fingadar, 2020-06-11:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

rmackay9, 2020-06-12:
If we did this though there would be no way for the pilot to accept or reject the new tune values.  At the moment the pilot accepts the new tune values by landing and disarming in AutoTune mode.

My guess is that the solution is actually to build a better interface between the GCS and autopilot so that the user has a very clear status report of the progress of the tune and the final tuning values along with an accept/reject button and a way to easy switch between the old and new tuning gains.

rmackay9, 2020-06-12:
I've changed the title to reflect the request a little more clearly.

fingadar, 2020-06-12:
The problem (for me) is, that in the case of large copters the new behavior is a litte bit annoying. Especially because I can more or less confirm the issue #11478 . 

Even when I‘m starting Autotune in 4.03 from Loiter, during Autotune (I’m using a switch) the copter behaves more or less like being in AltHold mode. Definitely not comparable to Loiter, even after successfully Autotune procedure ends.
Always landing big copters in a flight mode that feels like AltHold doesn’t make fun even with little wind.

In 3.x, Autotune was a very simple thing. Now, it is much more complicated with the chance of damage at landing in my opinion.

IamPete1, 2020-06-12:
The only requirement for saving gains is to be in AutoTune when disarming, you can land in whatever mode you like, switch to AutoTune and disarm. Any time you switch out of AutoTune the new gains are not lost you can switch back into AutoTune to try them out.

Maybe a simple solution would be to print the gains before discarding so people can easy retrieve them if they discarded them in error.

mboland, 2020-06-12:
Could we alleviate this with a Wiki update?
I am seeing this the most where people setup autotune as a flight mode or use another flight mode to land rather than having it on a seperate switch.
The operation has always been, to me anyway, super simple.
Just have it on a seperate switch so after the autotune you can switch in and out of your new settings, and if you don’t like them stay out of autotune when you disarm, or conversely, if you like the new tune stay in autotune when disarming and they will be saved.
Please correct me if I am wrong.

Hwurzburg, 2020-06-23:
[image]

andyp1per, 2020-06-23:
Note that I think there is also a bug in autotune currently. If you land and disarm with autotune enabled it will save the gains. If you then disable autotune it will re-instate the old gains but not save them. So as long a you don't then save through your GCS you should be ok, but it's highly confusing.

IamPete1, 2020-06-23:
@Hwurzburg I think that is only a good idea if your have finished auto tune but not if you got half way through. The gains its trying are only used for a step input so there is no guarantee they will be safe for normal flight.

Hwurzburg, 2020-06-23:
@IamPete1  However, you certainly would have known that before you landed..I certainly would not restart with an autotune that left the copter unstable...I dont anyone would....but it works well if its not at an ustable point....some copters/VTOLs I have had cant get thru one axis in a single tune session due to battery limitations....those would never complete an AUTOTUNE

Hwurzburg, 2020-06-23:
@andyp1per we can note that in the wiki, or fix in the code....I can probably do the wiki faster than anyone will volunteer to fix the code....or will they?

IamPete1, 2020-06-23:
@Hwurzburg My understanding is that you can't tell if its unstable because its not using the new gains all the time. For the hovering between step inputs it still uses the original gains. The autotune gains are only used for the step input and recovery then it switches back to reposition. I have not checked the code, but @lthall advised me against recommending people do that. 

Having said that I have totally done it, in most cases its fine, you just have to re-revalue the tune before re-starting autotune to ensure it will still be able to recover.

andyp1per, 2020-06-23:
@Hwurzburg I'm happy to fix the code

lthall, 2020-06-23:
Andy with my help is looking at fixing this.

Just to clear.
Autotune should save gains after it has completed successfully.
If autotune is then restarted it should use the saved gains not the old gains.
The Autotune gains should only ever save if 
- you have initiated Autotune on the mode switch and you land in autotune mode.
- you have initiated Autotune using the chX switch and you have landed in the high state.
If using the chX option you can only ever test autotune gains after autotune has successfully completed.

Something that people sometimes don't understand. Autotune does not gradually improve the tune. It deliberately moves into dangerous and unflyable tunes to make measurements to find the optimal tune. If you use autotune values from part way through the tune you may get an aircraft the is very poorly tuned or very over tuned and oscillating.

lthall, 2020-06-23:
Another point, using gains from some random point does not speed up the tune unless you were dangerously far away from a reasonable tune to start with.

If you started autotune from a good tune and stopped just before a successful tune. Then restarted the autotune using those values, the new tune would take just as long. This is because autotune has to find a low P value tune, a high P value tune, a Low D value tune and a High D value tune then it can zero in on the optimal tune. If you restart it has to find these again.

Hwurzburg, 2020-06-23:
I will change it  and remove the last comment...

andyp1per, 2020-06-25:
https://github.com/ArduPilot/ardupilot/pull/14680

IamPete1, 2021-08-16:
I think we can probably close this? I think in the end we kept the code the same and improved the wiki.

closed 2021-08-17 by rmackay9