ArduPilot/ardupilot #14590: Mission Plans - Add DO_SET_SPRAY - to enable and desable spray in mission

framosnt, 2020-06-12:
## Feature request

the sprayer drones In mission need to turn on and off the spray system, like photography SURVEY GRID, for this, we need new COMMAND "DO_SET_SPRAY" for use with SPRAY_PARAMETER for variable spray control. 

**Is your feature request related to a problem? Please describe.**
When drone will do turns to next strip, we need to turn off using RCx_Function = 15, but, I need to do this by command line.

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

tatsuy, 2020-09-22:
I think that this is achieved by #14736 .
It has not been merged yet.

amilcarlucas, 2021-02-18:
Got merged.

closed 2021-02-18 by amilcarlucas