ArduPilot/ardupilot #14593: Spline waypoints can break when target==destination

peterbarker, 2020-06-13:
## Bug report

**Issue details**

We have an autotest which tests the spline waypoint code.   It has started to fail more often in CI now, so we should probably look at it.

The peculiarity in the test is that there's a spline waypoint with location (0,0,0), indicating that the current location should be used.  (as an aside, this should probably really mean the waypoint is ignored - that would require some interesting restructuring in Copter's auto mode to skip that waypoint in `do_spline_start`).

What seems to happen (at random) is that the vehicle is at the waypoint, but the `target_velocity_length` is non-zero - so we don't set is_reached_destination - and _spline_vel_scaler ends up as zero so we don't make headway anywhere.

The failure happens at random.  Previously rather seldom, now rather more.  Some recent changes in our SITL simulation may be provoking this more.

 I'm invoking it like this:

```
./Tools/autotest/autotest.py --gdb --debug build.Helicopter 
```

followed (releatedly) by

```
./Tools/autotest/autotest.py --gdb --debug test.Helicopter.SplineWaypoint -B AC_WPNav::advance_spline_target_along_track --disable-break
```

That last step I had to run 4 times before provoking the bug on master.

I'm using this patch to drop into the debugger to inspect things:
```
pbarker@bluebottle:~/rc/ardupilot(master)$ git diff
diff --git a/Tools/autotest/arducopter.py b/Tools/autotest/arducopter.py
index ceb7e0cf97..d4dde60bfe 100644
--- a/Tools/autotest/arducopter.py
+++ b/Tools/autotest/arducopter.py
@@ -5233,6 +5233,7 @@ class AutoTestHeli(AutoTestCopter):
         tstart = self.get_sim_time()
         while True:
             if self.get_sim_time() - tstart > timeout:
+                self.send_debug_trap()
                 raise AutoTestTimeoutException("Vehicle did not disarm after mission")
             if not self.armed():
                 break
```

Additionally, this line (currently AC_WPNav:898):
```
        _pos_delta_unit = target_vel / target_vel_length;
```

really deserves  a comment at the very least - because.... what?!

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Heli

**Hardware type**
SITL

peterbarker, 2021-03-11:
This might be fixed by https://github.com/ArduPilot/ardupilot/pull/15896

lthall, 2021-05-04:
Fixed

closed 2021-05-04 by lthall