ArduPilot/ardupilot #14594: No dataflash-over-mavlink on telemetry ports

stephendade, 2020-06-13:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

The dataflash-over-mavlink logging (LOG_BACKEND_TYPE == 3, for example) does not provide logging messages over the telemetry ports. It works fine on USB though.

I tried powering via power module (instead of via USB) with no change.

Note this was discovered as part of testing my [Rpanion-server](https://github.com/stephendade/Rpanion-server) software.

**Version**
Current master

**Platform**
[X  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
Tested on a Pixhawk1

**Logs**
N/A

peterbarker, 2020-06-13:
I've just tested on a PixRacer here and it's working fine for me.

stephendade, 2020-06-14:
After much debugging, turns out the issue is with mavlink-router. Turns out it won't route packets to the FC unless the target sys/comp ID match exactly. Which doesn't work if I'm using MAV_COMP_ID_ALL or MAV_COMP_ID_LOG.

Not sure why this _was_ working on a direct USB connection in the first place.

Serves as a notice for anyone else using mavlink-router and datafalsh logging.

closed 2020-06-14 by stephendade