ArduPilot/ardupilot #14596: Tools/AP_Bootloader/support.cpp:95:21: error: comparison of unsigned expression < 0 is always false [-Werror=type-limits]      while (reserved < FLASH_RESERVE_END_KB * 1024U)

pratik206, 2020-06-13:
hello,

I am getting this error while building the bootloader for mRoControlZeroF7 device. Please help me.

[81/85] Linking build/mRoControlZeroF7/Tools/AP_Bootloader/liblibcanard.a
../../Tools/AP_Bootloader/support.cpp: In function 'void flash_init()':
../../Tools/AP_Bootloader/support.cpp:95:21: error: comparison of unsigned expression < 0 is always false [-Werror=type-limits]
     while (reserved < FLASH_RESERVE_END_KB * 1024U) {
                     ^
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors

Waf: Leaving directory `/root/GitHub1/ardupilot/build/mRoControlZeroF7'
Build failed
 -> task in 'bootloader/AP_Bootloader' failed (exit status 1):
        {task 139668615409008: cxx support.cpp -> support.cpp.2.o}
 (run with -v to display more information)

peterbarker, 2020-06-13:
On Sat, 13 Jun 2020, pratik206 wrote:

> [81/85] Linking build/mRoControlZeroF7/Tools/AP_Bootloader/liblibcanard.a
> ../../Tools/AP_Bootloader/support.cpp: In function 'void flash_init()':
> ../../Tools/AP_Bootloader/support.cpp:95:21: error: [...]

Which compiler are you using?

pratik206, 2020-06-13:
I am using gcc-arm-none-eabi-6-2017-q2

pratik206, 2020-06-13:
Its working just by commenting that lines.

closed 2020-06-13 by pratik206

peterbarker, 2020-06-13:
On Sat, 13 Jun 2020, pratik206 wrote:

> I am using gcc-arm-none-eabi-6-2017-q2

Well that's odd.  I'm using that compiler as well and can't provoke the 
problem.