ArduPilot/ardupilot #14606: Watchdog Messages with Omnibus Nano V6 and latest master

anbello, 2020-06-15:
**Issue details**
I built ArduCopter from master (82480f0a3d) for Omnibus Nano V6 after adding
`define HAL_WITH_DSP 1`
to hwdef.dat to have dynamic FFT, I already did this with past version without any problems.

After reboot (sometimes), I see messages relative to watchdog that resolves after next reboot, then the copter works (and fly) with no problems.

Follows an extract of messages on GCS and (attached) a tlog.
```

[MAV 001:1] SUCCESS: Executed CMD: 246
[MAV 001:1] Found corrupt log 7 at 0x5471, erasing
[MAV 001:1] Calibrating barometer
[MAV 001:1] Initialising ArduPilot
[MAV 001:1] Barometer 1 calibration complete
[MAV 001:1] INS: alloc 6144 bytes for ISB (free=64144)
[MAV 001:1] EKF3 IMU0 buffs IMU=6 OBS=2 OF=5 EN:5, dt=0.0120
[MAV 001:1] EKF3 IMU0 initialised
[MAV 001:1] EKF3 IMU0 tilt alignment complete
[MAV 001:1] RCInput: decoding IBUS
[MAV 001:1] Found 7 complete logs at 0x0001-0x5470
[MAV 001:1] Baro: skipping calibration after WDG reset
[MAV 001:1] INS: alloc 6144 bytes for ISB (free=64144)
[MAV 001:1] Restored watchdog attitude -1 0 -0
[MAV 001:1] EKF3 IMU0 buffs IMU=6 OBS=2 OF=5 EN:5, dt=0.0120
[MAV 001:1] EKF3 IMU0 initialised
[MAV 001:1] EKF3 IMU0 tilt alignment complete
[MAV 001:1] WDG: T6 SL419 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0
[MAV 001:1] IE32768 IEC13 TN:
[MAV 001:1] PreArm: Internal errors 0x800 l:225 ,watchdog_rst
[MAV 001:1] WDG: T6 SL419 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0
[MAV 001:1] IE32768 IEC13 TN:
[MAV 001:1] WDG: T6 SL419 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0
[MAV 001:1] IE32768 IEC13 TN:
[MAV 001:1] WDG: T6 SL419 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0
[MAV 001:1] IE32768 IEC13 TN:
[MAV 001:1] PreArm: Internal errors 0x800 l:225 ,watchdog_rst
[MAV 001:1] WDG: T6 SL419 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0
[MAV 001:1] IE32768 IEC13 TN:
[MAV 001:1] SUCCESS: Executed CMD: 246
```

**Version**
 ArduCopter V4.1.0-dev (82480f0a)

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X quadcopter

**Hardware type**
Omnibus Nano V6

**Logs**
[2020-06-14 16-10-48.tlog.zip](https://github.com/ArduPilot/ardupilot/files/4778526/2020-06-14.16-10-48.tlog.zip)

anbello, 2020-06-16:
I did other test with master from yesterday (7f42930) and I see no more WDG messages, maybe this commit form @tridge https://github.com/ArduPilot/ardupilot/commit/7f429302ed18634fcf8846a098fc2f887690772e solved the issue.

rmackay9, 2020-06-17:
@anbello, well, that would be great if this has fixed it.  This fix you mention is included in Copter-4.0.4-rc2.

tridge, 2020-06-17:
I suspect it is just that your dataflash is fully erased now, so you're not getting the delays. I doubt my change fixed it

peterbarker, 2020-06-17:
@anbello you could try setting `LOG_DISARMED` and waiting for stuff to fill up and then doing a log-erase; that should trigger the watchdog.

anbello, 2020-06-17:
> @anbello you could try setting `LOG_DISARMED` and waiting for stuff to fill up and then doing a log-erase; that should trigger the watchdog.

I will do asap.
How can I know when the flash is full? Will I see a message?

andyp1per, 2020-06-17:
You won't see a message - the log will just wrap. You will know when listing the logs gives you a single log of 16Mb. You can speed this up by setting your LOG_BITMASK to everything

anbello, 2020-06-17:
I did as @peterbarker said above, for two times, and I see no watchdog messages.
It seems all OK.

peterbarker, 2020-11-17:
Thanks @anbello

closed 2020-11-17 by peterbarker