ArduPilot/ardupilot #14610: Body rate control in GUIDED mode.

lmark1, 2020-06-15:
## Feature request

I have already described the problem [here](https://discuss.ardupilot.org/t/mavros-body-rate-control/57427). In short, I would like for the Copter firmware to support [body_rate, thrust] setpoints in GUIDED mode. 

To be more precise, I would like for the Copter firmware to accept  the following [SET_ATTITUDE_TARGET](https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html#set-attitude-target) type_masks:
- 0 - Ignore none of the setpoints
- 8 - Ignore the orientation setpoint

Copter currently only takes into account these type_masks:
- 7 - Ignore body rate setpoints
- 4 - Ignore body roll and pitch rate setpoints

If there is no alternative to solving this issue, I am willing to help make appropriate changes to the GUIDED mode in the Copter firmware to make this possible.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

acxz, 2020-11-13:
Just piping in to say I would love this feature!

atharva417, 2021-08-08:
Hello @lmark1
I'm working on a project where I want to publish [body_rate, thrust] setpoints in GUIDED mode. Were you able to develop the support for the Copter firmware?
If yes, can you please share the reference?

rmackay9, 2021-08-08:
I somehow missed this issue when it was first raised.  I'm afraid that AP's guided mode does not have a rate controller option.  This would essentially be the same as controlling the vehicle in Acro mode.

Copter-4.1's Guided mode has these controllers:
    - takeoff (to a specified altitude)
    - position/waypoint (fly to a specified location in 3d)
    - acceleration (accelerates the vehicle in a give 3D direction)
    - velocity (flies at the given 3D velocity)
    - position/velocity/accel (various combinations of position, velocity and accel)
    - angle + thrust or climb rate (flies at the given lean angle and thrust or climb rate)

We have all the tools to add a rate controller so we should be able to do it.

atharva417, 2021-08-09:
> We have all the tools to add a rate controller so we should be able to make this enhancement. If someone does a PR that would speed things along of course and would be greatly appreciated but in either case hopefully we will get to this eventually.

Glad to hear that! Looking forward to this enhancement.

lmark1, 2021-08-09:
Hello,

I made some changes to add the body-rate control [here](https://github.com/larics/ardupilot/commits/body-rate-control) for Copter v3.6 iirc.
I will try to clean it up, apply it on Copter v4.1 and make a pull request.

rmackay9, 2021-08-10:
@lmark1,

Thanks for that.

atharva417, 2021-08-10:
It worked for me.
Thank You so much!!!! @lmark1

lmark1, 2021-08-14:
Hello,

I opened a PR  #18327 for this issue.

shubham-shahh, 2021-10-07:
> Hello,
> 
> I made some changes to add the body-rate control [here](https://github.com/larics/ardupilot/commits/body-rate-control) for Copter v3.6 iirc. I will try to clean it up, apply it on Copter v4.1 and make a pull request.

will this work with [mavros_controllers](https://github.com/Jaeyoung-Lim/mavros_controllers)

hendjoshsr71, 2022-01-12:
@lmark1  I'm closing this one as we are about to merge  https://github.com/ArduPilot/ardupilot/pull/18833 instead.

Thank you again for this one it was quite good! It just so happened the maintainer of copter controls came along to fill this out completely.

If you would like to give it a test for us we would love that!

closed 2022-01-12 by hendjoshsr71