ArduPilot/ardupilot #14617: AP_Scripting Logging example not working on Pixhawk1 FC

Notorious811, 2020-06-16:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED 
## Bug report

**Issue details**

I tried to run lua scripting logging.lua example on Pixhawk1 v2.4.8 FC after building and flashing the firmware using master branch of Ardupilot, that example runs fine and also create the AHRS_DATA.csv file but it does able to write the timestamp, roll, pitch and yaw values in the AHRS_DATA.csv file. This logging example runs fine on SITL, just not working on actual flight controller.

**Version**
ArduCopter V4.1.0-dev (042ca786)

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Pixhawk1 v2.4.8

IamPete1, 2020-06-17:
Its the message in the dataflash log working as expected?

Notorious811, 2020-06-18:
> Its the message in the dataflash log working as expected?

yup, dataflash log working fine on pixhawk, after observing dataflash log under SCR label created from lua script it looks much more smooth than actual log of AHRS data (roll,pitch,yaw) under ATT label

IamPete1, 2020-09-23:
I'm not sure this is fixed, that script is just written in such a way as to get round the bug.

I suspect the issue is that leaving a file open between scripting calls causes it to get cleared in some way, or maybe its just if there is no activity in the file for a certain time.

There must be some difference in file handling between the SITL and real file-system. Such a difference flagged in https://github.com/ArduPilot/ardupilot/issues/15277 no doubt there are some others.

I have been meaning to write a script to try out leaving a file open for long and longer time periods, and over function calls.

Notorious811, 2020-09-23:
I closed it cause i am able to write files using AP_Filesystem now, i was facing similar issue with even with AP_Filesystem earlier that i was not able to write but later i found that SD card format was not fatfs, i reformatted in fatfs and then able to write successfully with AP_Filesystem, that's why i thought it might be similar case with scripting too, that's why i closed this issue, but i haven't tested it yet with lua scripting

Notorious811, 2020-09-23:
tested that PR #15234 will fix this issue on real hardware, closing the issue for now, thanks

IamPete1, 2020-09-23:
IMHO the PR is a work around not a fix, this is still a issue. Although it would be nice to frame the exact circumstances the causes it.

Notorious811, 2020-09-24:
Aye aye sir, will try to find causes of this issue

IamPete1, 2020-09-28:
I have also found that opening a file in `"r+"` mode does not work in SITL I would guess also not on real hardware but not tested.

IamPete1, 2020-10-20:
Done some more investigation on this during the devcall. Files seems to work as expected in a single function call, or in a 0 time call back `return update 0`. Testing with this patch shows that AP_FS seems to think the files writes are working as expected.

https://github.com/IamPete1/ardupilot/commit/7dda2c40c1ab89fa7dab95a977972a6f444e0cfc

Symptoms are the same as above, we get a file with the correct name but of 0 size, both when viewed via MAVFTP and directly on the SD card.

PR #15769 (2020-11-11): AP_Filesystem: fixed flushing with posix compat layer
this ensures the directory entry in the FAT is updated on fflush()

Fixes #14617
commit 0db87453 (2020-11-11): AP_Filesystem: fixed flushing with posix compat layer

this ensures the directory entry in the FAT is updated on fflush()

Fixes #14617
closed 2020-11-11 by tridge