ArduPilot/ardupilot #14625: Message Mismatch between Mavros and ArduPlane

LightningPORTO, 2020-06-17:
## Bug report

**Issue details**

The /mavros/setpoint_position/global has the message type " geographic_msgs/GeoPoseStamped ", however, the ArduPlane firmware requests the same mavlink message ( SET_POSITION_TARGET_GLOBAL_INT (86) ) to have the structure as " mavros_msgs/GlobalPositionTarget " .

So arduplane specifically requests " mavros_msgs/GlobalPositionTarget " but mavros only sends " geographic_msgs/GeoPoseStamped " on the same topic.

This issue has also been opened on the Mavros github: https://github.com/mavlink/mavros/issues/1441

Any work arrounds / suggestions?

**Version**
Master - ArduPlane 4.0.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane

**Hardware type**
SITL

IamPete1, 2021-08-08:
Looks like the mavros issue has been close, does that mean we can close this? I'm not really sure what we can do about it in AP anyway?

LightningPORTO, 2021-08-08:
Yes, I'm closing it now

closed 2021-08-08 by LightningPORTO