ArduPilot/ardupilot #14626: RCInput not correctly identifying ST24 serial protocol

JoshG20, 2020-06-17:
## Bug report

**Issue details**

When the ST24 receiver, bound to a ST16 transmitter, is plugged into the SPKT/DSM port, it is incorrectly identified as a DSM receiver. The decoder gets the number of channels correct (12) but the decoded result is randomly fluctuating inputs because the wrong decoder is being used. I tested PPM to make sure it works, and it takes over a minute to detect the switch both ways. My guess is that the ST24 decoder has an issue and it's choosing DSM only because of a random chance packet that satisfies the selection requirements.

**Version**
This is on the latest stable version.

**Platform**
[     ] All
[     ] AntennaTracker
[ **X** ] Copter
[     ] Plane
[     ] Rover
[     ] Submarine

**Airframe type**
Quad X

**Hardware type**
Pixhawk 1

**Logs**
_Not useful, RC inputs are all over the place_

andyp1per, 2020-06-17:
Are you able to try https://github.com/ArduPilot/ardupilot/pull/14174 to see if it helps? It more precisely decodes DSM which may mean that ST24 works correctly

JoshG20, 2020-06-17:
Can I use your branch directly or should I cherry pick the commits to a stable branch?

andyp1per, 2020-06-17:
The branch is off master - should be fine. I can re-base if it will help.

JoshG20, 2020-06-17:
No as long as there's no major issues that's fine with me. setting up cygwin now, only have the pixhawk stuff installed atm.

andyp1per, 2020-06-17:
I'm happy to do you a build if you tell me what platform you need

JoshG20, 2020-06-17:
That would be great! What else do you need beyond the things in the bug report?

andyp1per, 2020-06-17:
https://www.dropbox.com/s/stekckg786oj6qj/arducopter.apj?dl=0

JoshG20, 2020-06-17:
Thanks, ill give it a try!

JoshG20, 2020-06-17:
Unfortunately the same issue persists. It selected DSM again and the sticks still move wildly with no physical movements, and I know the receiver is outputting clean packets because it works with pixhawk at least in hitl.

JoshG20, 2020-06-17:
The pixhawk uses similar decoder code so my suspicion is that the issue is where it selects the decoder. Is there any way to force it to use the st24 decoder to see if it works correctly?

andyp1per, 2020-06-17:
Not easily without modifying the code.

JoshG20, 2020-06-17:
The st16 transmitter and st24 receiver have been surprisingly incompatible considering that someone mustve needed it if they wrote them into pixhawk a while ago, I guess they don't use it anymore. I wish it worked properly, this thing is a great piece of kit, especially with one of the yuneec cams.

lipper-chengdu, 2023-05-10:
I know where the problem is and I will fix it

> The st16 transmitter and st24 receiver have been surprisingly incompatible considering that someone mustve needed it if they wrote them into pixhawk a while ago, I guess they don't use it anymore. I wish it worked properly, this thing is a great piece of kit, especially with one of the yuneec cams.
I know where the problem is and I will fix it

JoshG20, 2023-05-10:
Apologies for not replying to this in a long time. I tried the sr24 receiver on a brand new pixhawk 4 and it worked perfectly. It may be an issue exclusive to the radiolink pixhawk 1.

closed 2023-05-10 by JoshG20

romucifer, 2024-01-10:
> Je sais où est le problème et je vais le résoudre
> 
> > L'émetteur st16 et le récepteur st24 ont été étonnamment incompatibles étant donné que quelqu'un a dû en avoir besoin s'il les avait écrits dans pixhawk il y a quelque temps, je suppose qu'ils ne l'utilisent plus. J'aimerais que cela fonctionne correctement, cette chose est un excellent kit, surtout avec l'une des caméras yuneec. 
> > Je sais où est le problème et je vais le résoudre

Hello this st24 work ?

lipper-chengdu, 2024-01-15:
> Hello this st24 work ?

The repair code has been submitted to the MASTER , but the latest version does not include this code yet, and I don't know why.The version I built locally has been tested and is working properly.
https://github.com/ArduPilot/ardupilot/pull/23742