ArduPilot/ardupilot #14633: Problems with Attitude Rates Reported by MAVlink Message 30 (ATTITUDE)

cheintz, 2020-06-18:
## Bug report

The numerical derivative of the yaw reported by MAVlink is not equal to the yawspeed reported in the same message. Instead, the derivative of the reported yaw angle is equal to (approximately) 1/cos( roll)   * yawspeed.  See figure below. 

I tested this in FBWA at trim throttle and with pitch nominally zero. I suspect there's an issue with ambiguity over which frame (body vs. earth)  the rates are reported in. There's something wrong with rollspeed, as well,  but it is less obvious to me.

This plot is a log of output obtained via MAVlink through DroneKit. 

[image]

I suspect this may be causing some of the issues I have been troubleshooting in SITL with respect to poorly coordinated turns and a discrepancy with an algebraic manipulation of the "lateral acceleration to roll" function https://github.com/ArduPilot/ardupilot/blob/4aa10a6c30acabdb395ea42643f1e97b21113ac6/libraries/AP_L1_Control/AP_L1_Control.cpp#L79

**Version**
Plane 3.9.0 and 4.0.5 at a minimum

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL default in Ubuntu (not using jsbsim or anything else)

**Hardware type**

STITL in Ubuntu

**Logs**

[yawSweep405.zip](https://github.com/ArduPilot/ardupilot/files/4801402/yawSweep405.zip)

cheintz, 2020-06-25:
Further analysis of my logs indicates that reported values are very likely the body-fixed angular velocities, where the forward, right, and vertical body frame velocities are reported as the roll, pitch, and yaw velocities, respectively. This makes a lot of sense, but could probably be a documentation change, rather than a code/function/implementation change.

IamPete1, 2021-08-08:
Sounds like this is just due to the different reference frames.

closed 2021-08-08 by IamPete1