ArduPilot/ardupilot #14640: PreArm: Compass not healthy when using dual-antenna GPS (with HDT) and EKF3

phamelin, 2020-06-19:
## Bug report

**Issue details**

I'm using a dual-antenna GPS (Trimble MB-Two) which provide true heading data through HDT NMEA messages. All compass have been disabled. Arming fails with the following message:

`PreArm: Compass not healthy`

The problem can be reproduced easily using SITL with the following parameters:

```
# NMEA DGPS simulation
SIM_GPS_TYPE     5.000000
SIM_GPS_HDG      1.000000
GPS_TYPE         5.000000
GPS_HDOP_GOOD    220.000000

# Use EKF 3 with DGPS for heading
COMPASS_ENABLE   0.000000
COMPASS_USE      0.000000
COMPASS_USE2     0.000000
COMPASS_USE3     0.000000
EK2_ENABLE       0.000000
EK3_ENABLE       1.000000
AHRS_EKF_TYPE    3.000000
EK3_MAG_CAL      5.000000
```
This issue was already resolved for EKF2 (https://github.com/ArduPilot/ardupilot/issues/5454), but since I need to use a dual-antenna GPS for heading I must use EKF3.

The problem arises from this code in `ArduCopter/AP_Arming.cpp` :

```
#ifndef ALLOW_ARM_NO_COMPASS
    // if external source of heading is available, we can skip compass health check
    if (!ahrs.is_ext_nav_used_for_yaw()) {
        const Compass &_compass = AP::compass();
        // check compass health
        if (!_compass.healthy()) {
            check_failed(true, "Compass not healthy");
            return false;
        }
    }
#endif
```

with `ahrs.is_ext_nav_used_for_yaw()` defined as follow in `libraries/AP_AHRS/AP_AHRS_NavEKF.cpp`: 

```
bool AP_AHRS_NavEKF::is_ext_nav_used_for_yaw(void) const
{
    switch (active_EKF_type()) {
    case EKF_TYPE2:
        return EKF2.isExtNavUsedForYaw();
    default:

        return false; /// <-- ALWAYS FALSE FOR EKF3
    }
}
```

I think that `AP_AHRS_NavEKF::is_ext_nav_used_for_yaw()` should be replaced by a more generic function that would check if a compass is used.

**Version**
4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
SITL and Cube Black

**Logs**
None

phamelin, 2020-06-29:
It looks like it has already been solved in master (https://github.com/ArduPilot/ardupilot/commit/a66479104de96d42d41ce231d414d853232a485a).

closed 2020-06-29 by phamelin

Alondruck, 2020-09-15:
@phamelin 

Hello Phillipe, I wanted to tell you that I am entering the world of navigation controllers for autonomous vehicles.
I want to consult with you on an issue that I’m having, I’m trying to integrate the Trimble MB-Two in my system using the Pixhawk4 controller with ardupilot-ardurover 4.1.0 firmware. As of now, I managed to set up the Trimble to output the position and heading messages (Check it by connecting one of the serial port of the Trimble MB-Two to the computer and viewing the message through a serial terminal), but I still didn't manage to set up the Pixhawk correctly to use this data.   
 
**The status right now:**

- The Pixhawk does not recognize the message provided via serial by the MB-Two. The Mission Planner shows that there is no connection to the second GPS.
- I also tried to put the parameter GPS_TYPE_2 = 1 which means that it automatically searches for GPS type but it was not able to identify the connection either.

**Ardurover parameters configuration:**
 
- GPS_TYPE2 = 5: Receive NMEA data from GPS 2
- SERIAL4_BAUD = 38: Baud rate = 38400 (configure same baud rate on MB-two side).
- SERIAL4_PROTOCOL = 20: This will enable Serial4 port to be used as NMEA output.
- GPS_AUTO_SWITCH = 1: Autopilot will automatically switch between the two GPS receivers picking the one with a better solution.
- EK2_ENABLE = 0: Disable EKF2.
- EK3_ENABLE = 1: This enables EKF3.
- AHRS_EKF_TYPE = 3: This control which NavEKF Kalman filter version is used for attitude and position estimation.
- EK3_MAG_CAL = 5: Enable external yaw.

**MB-Two configuration:**

[image]

- Port B is the one connected to the Pixhawk through the UART & I2CB pin on the Pixhawk board.
- NMEA GGA and HDT message is sent to Pixhawk.

Have a good day, regards.
 
**Version**
4.1.0

**Platform**
[ ] All
[ ] AntennaTracker
[] Copter
[ ] Plane
[X] Rover
[ ] Submarine

**Hardware type**
Pixhawk 4

**Logs**
None

phamelin, 2020-09-16:
@Alondruck I really don't know why this isn't working on your side. I've only used GPS #1, so maybe the problem comes from the fact that you are using the Trimble as the secondary GPS. Also, I think that you also need the RMC NMEA message.

sujiv0, 2021-04-06:
@Alondruck Have you able to solve the data not coming to Pixhawk issue?

sujiv0, 2021-04-07:
@phamelin Hi, I'm trying to use dual-antenna GPS (Trimble MB-Two) with Pixhawk2.1 CubeBlack for Heading. 
I checked the NMEA messages using a TTL-USB conveter in my Laptop, and all are receiving.
But when I'm connecting to Pixhawk, I'm not able to find any messages from Pixhawk related to the setup.

So, if you don't mind, please share the parameter for Pixhawk and Message setup for Trimble for this configuration.

Thanks in advance.

phamelin, 2021-04-12:
@sujiv0 As far as I know, @Alondruck already mentioned the critical parameters. Mine are:
```
GPS_TYPE         5.000000
COMPASS_USE      0.000000
COMPASS_USE2     0.000000
COMPASS_USE3     0.000000
EK2_ENABLE       0.000000
EK3_ENABLE       1.000000
AHRS_EKF_TYPE    3.000000
EK3_MAG_CAL      5.000000
```
On the Trimble side, I have enabled RMC, GGA and HDT. Trimble THS message will also be supported in a future release (see https://github.com/ArduPilot/ardupilot/pull/14762).