ArduPilot/ardupilot #14643: MAVFTP usability issues when combined with MP

rmackay9, 2020-06-20:
I am unsure if these are Mission Planner issues or flight code issues so I am documenting them both here and in the MP issues list: https://github.com/ArduPilot/MissionPlanner/issues/2413

All of these issues can be reproduced by doing the following:

- Open Mission Planner
- Open the Simulation page
- Ensure the "wipe" checkbox is checked
- Press the Copter icon at the bottom
- Open Config, MAVFtp

When trying to upload a Lua script to a copter in MP's simulator I've run into these problems:

- Click, "@ROMFS", then click on the unnamed (top directory?), 5 second delay before the directory listing is displayed.  5sec is not very long but it is very noticeably longer than if I click on one of the sub directories (i.e. logs) and then switch back to the top directory
[image]
- Attempting to create a directory with the same name as an existing directory causes MP to lock-up for 30 seconds.  The directory is not created and no error message is displayed
- Attempting to delete an empty directory causes MP to lock-up for 30 seconds.  A "Failed to delete file" message is displayed and the directory is not deleted

tridge, 2020-06-20:
The ROMFS filesystem is read-only, creating files there doesn't make any sense. The files get created as part of the firmware build process

rmackay9, 2020-06-20:
@tridge, ok thanks for the info.  I'm sure most people won't know this so we will need to protect them from their GCS locking up for 30sec.  I wonder if it's AP misbehaving and not providing a failure message or if it's MP not processing the failure.

@meee1 fyi

davidbuzz, 2020-06-20:
RO stands for Read Only.

rmackay9, 2020-06-22:
@davidbuzz, yes, that's understood and great but the GCS shouldn't lock up for 30seconds if the user clicks on the ROMFS directory.

rmackay9, 2020-06-25:
MP issues now resolved so closing, txs!

closed 2020-06-25 by rmackay9