ArduPilot/ardupilot #14644: Scripting: "Lua: open directory (@ROMFS/scripts) failed" message

rmackay9, 2020-06-20:
Recently it seems a new innocuous error message, "Lua: open directory (@ROMFS/scripts) failed", is being sent from the vehicle to the ground station.  There are apparently two places where scripts can be held (@ROMFS/scripts and APM/scripts) so we should only display the error message if we don't find either directory.

[image]

As a general rule we should try not to spam the user with messages the may appear scary (i.e. "failed" sounds bad) when they do not actually need to act upon them.

WickedShell, 2020-06-24:
@rmackay9 Do we want to change this to be "Lua: No scripts found in @ROMFS/scripts" or just suppress it in this case? The message comes from the same place we normally use to load scripts from the SD card folder, so it is nice to have a message of some form to help make it easier to tell what went wrong there. Otherwise we could pass a "fail silently" flag in to say for ROMFS that we don't care if the directory isn't there.

rmackay9, 2020-06-25:
@WickedShell,

Displaying a message if there are no scripts found in either of the two possible directories sounds like a good idea but I don't think we should display any message if there are no scripts in the ROMFS directory but there are scripts in the SD card directory.

peterbarker, 2020-07-02:
@rmackay9 so do we need to change the issue title now?

I believe the original issue was satisfactorily addressed.

rmackay9, 2020-07-02:
@peterbarker, yes, I think this is fixed now, txs!

rmackay9, 2020-07-16:
I thought this was fixed but it doesn't appear to be actually.  I'm still seeing a, "Lua: open directory (@ROMFS/scripts) failed" message each time I start SITL if SCR_ENABLE = 1.  Even if I have scripts in the scripts directory this message appears.

WickedShell, 2020-10-28:
The linked PR didn't actually address this, this is still pending.

IamPete1, 2024-05-04:
fixed by https://github.com/ArduPilot/ardupilot/pull/26312