ArduPilot/ardupilot #14653: MAVFTP does not work on Pixhawk1 boards but capabilities bit is still set

rmackay9, 2020-06-22:
I'm unsure if MAVFTP is suppose to work with a Pixhawk or not but it doesn't seem to at least not with the Mission Planner.  If it's not meant to work then the capabilities bit is being set incorrectly.

So to summarise there is at least one problem:

- MAVFTP doesn't work with Pixhawk1 (maybe it's shouldn't work?)

OR

- the MAV_PROTOCOL_CAPABILITY_FTP bit is being set incorrect for Pixhawk1 ([see code here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L4955))

This can be reproduced using latest Copter-4.1.0-dev on a Pixhawk1.  Attempt to connect with MP and you will find it stalls for 30seconds as it attempts (and fails) to load parameters using MAVFTP.  Then open MP's Config, MAVFtp screen and note that it's always blank (but the SD card is not blank)

khancyr, 2020-06-22:
Hello,

Two weeks ago I used Rover master on my pixhawk1-1M and mavftp was working with Mavproxy and Mission Planner. I will update this comment with the right Git Hash

kd0aij, 2020-08-23:
@rmackay9 How much free RAM do you have in your Pixhawk configuration?
@WickedShell Should GCS_MAVLINK::ftp_init send a warning when it fails?

WickedShell, 2020-08-24:
@kd0aij probably not a bad idea. Ideally we could pro grammatically nack it, but I don't see a good way to do that. As an extension to that we could also go out of our way to unflag the FTP capabilities bit.

Part of the problem with allocating things like the FTP at the end is not only do we need a fair amount of memory, we need it in 3 different contiguous chunks, it's possible you have enough RAM, but might not have enough in a contiguous chunk for something like the replies buffer to be allocated.

WickedShell, 2020-08-24:
A more aggressive work around to keep FTP working, but at the cost of slower response times would be to add code that says "if we couldn't allocate our space for 30 replies in the buffer, try cutting it down to 15 replies, then try 5 replies" etc. At some point though if we can't keep a minimum number of replies available it's not going to be responsive. It would help lower the memory requirement, but to really benefit from this you'd actually need to rearrange the order things get done in (allocate the thread first, have it wait until we signal it can actually start, then try and allocate the buffers, and on success signal the thread it can actually run now). THis is a larger rewrite, but would allow for more automatic scaling of buffer sizes.

IamPete1, 2021-08-16:
It seems this is just a factor of the amount of free memory, we now will send a warning. We could open a new issue (or rename this one) for FTP to use less memory?

kd0aij, 2021-08-16:
created new issue #18351

closed 2021-08-16 by kd0aij