ArduPilot/ardupilot #14661: Copter: GPS arming check fails when no GPS present despite good position from visual odometry

IamPete1, 2020-06-23:
It is not possible to arm into loiter with no GPS, however once armed the vehicle will fly in loiter perfectly with position from visual odometry.

Not sure if its a good idea to just disable the GPS checks if there is no GPS, hopefully any lack of position should be caught by EKF/Visual Odom/Mandatory GPS checks.

Would be nice if we could display good position from visual odom with green 'GPS' light too.

Maybe we should look to separate the 'do I know where I am' checks from the GPS checks, in the future there are only going to be more ways of getting position that are not GPS.

rmackay9, 2020-06-23:
@IamPete1, yes, I agree with all of this I think.  We should probably have different arming checks (and notification) based on whether the EKF has been configured to use the GPS or not.

anbello, 2020-06-23:
@IamPete1 I am able to arm in Loiter using visual odometry. Which value do you use for GPS_TYPE and  EKx_GPS_TYPE ?

IamPete1, 2020-06-23:
@anbello GPS_TYPE = 0, EK3_GPS_TYPE = 3, ARMING_CHECK = 1. I hit the low hdop pre arm. It would arm if I turned off the GPS arming checks, but I shouldn't have to. (I'm on Copter - dev)

anbello, 2020-06-23:
@IamPete1 OK you are right, I forgot to have turned off the GPS arming check in the copter I use for Visual Odometry test.

peterbarker, 2024-11-23:
Resolved by #22393

closed 2024-11-23 by peterbarker