ArduPilot/ardupilot #14683: EKF2: predicted pos-horiz-rel/abs switching true/false with no GPS or optical flow

rmackay9, 2020-06-25:
I'm seeing the "pred_pos_horiz_rel" and "pred_pos_horiz_abs" change rapidly between true and false when Copter-4.0.3 (or master) is loaded onto a CubeBlack with no GPS.

The vehicle has no source of horizontal position or velocity so I really can't imagine how these flags could ever be true.
[image]

rmackay9, 2020-06-25:
It's possible this is a MP bug although MP's mavlink inspector also shows the flags field changing rapidly so if it is an MP issue it's something lower down at the packet processing level.

rmackay9, 2020-06-25:
This turned out to be an MP issue so closing

closed 2020-06-25 by rmackay9

Nitrousfree, 2020-08-15:
@rmackay9  hi, I have this same issue while using optical flow without external GPS module. My pos_horiz_abs keep shows OFF. What happened is my quadcopter rise rapidly out of control while autonomous takeoff and land. The devices I'm using are rpi 3 and pix 2.4.8(arducopter px4 3.6.8).

Nitrousfree, 2020-08-15:
Is it really MP's problem or is because of pixhawk's version？ Because when I switch to Copter-4.0.0 above I see no way for me to set any parameters about ekf3. I'm currently using my drone indoor so optical flow and lidar are my position and attitude sensors. It  success to autonomous takeoff by rpi but after reaching the attitude(sonar range), the drone keep raise out of control force me to shutdown by rc controller.